lowRISC/ibex #957: internal gated clock running during reset

bp-ouhaha, 2020-06-15:
from a low-power point-of-view, having the main clock running before fetch_enable_i is set high to kick-start it, is not a great idea ;)

I've traced it up to ibex_controller.sv, where in RESET state, the ctrl_busy is still on default 1'b1. Could we maybe make it 1'b0 in reset state?

To make it work, I also add to activate the internal clock when fetch_enable_i is high, otherwise if fetch_enable_i is only 1 cycle long, you miss it.

GregAC, 2020-06-18:
We were wondering if `fetch_enable_i` was a useful signal to have altogether. It's a legacy from the PULP project roots of Ibex that may no longer have use. Our thoughts being that `fetch_enable_i` only has relevance immediately after reset and if you don't want the core to start fetching immediately after reset then just hold the core in reset for longer until you do want it to start fetching.

To put it another way the state of 'just out of reset, waiting to be asked to start fetching' is not a useful state. In which case we can just remove `fetch_enable_i` altogether.

If you do find it useful to have (as opposed to just holding rst_ni low for longer) a description of your use case would be handy, otherwise I think we'll just remove it avoiding this whole issue.

tomroberts-lowrisc, 2020-06-18:
I think we probably need to retain it for now since I think we are keeping Ibex compatible with PULP - @vogelpi will know.

vogelpi, 2020-06-18:
We just discussed the `fetch_enable_i` here in the lab. Most of our own but also other cores known in the group feature some mechanism to gate program execution after reset. This has for example proven useful when testing it on the chip tester (or to change the boot address after a reset but before the core starts running). The `fetch_enable_i` in Ibex is maybe not the cleanest solution, but it is safe to just wire it to 1'b1 if you don't need it. Newer cores like Ariane and internal cores use a memory mapped register that is read in the bootloader. If the value read is 0, the core goes into sleep like a WFI. Some other open-source cores use something like a fetch trigger/kick start for the same purpose.

Now for the original question: I am not sure if I fully understand. You want to disable the clock before setting the fetch enable, right? To this end, you are suggesting to set `ctrl_busy = 1'b0` in RESET (`ibex_controller.sv`, Line 398) but then factor `fetch_enable_i` into `clock_en` directly (`ibex_core.sv`, Line 357)? I don't see anything wrong with that.

bp-ouhaha, 2020-06-18:
indeed that was the suggestion (last part added in ibex_core.sv):
  assign clock_en     = core_busy_q | debug_req_i | irq_pending | irq_nm_i | fetch_enable_i;

in ibex_controller.sv (added else case):
        if (fetch_enable_i) begin
          ctrl_fsm_ns = BOOT_SET;
        end else begin
          ctrl_busy_o = 1'b0;
        end

vogelpi, 2020-06-19:
Good point! Of course the clock should be gated in WFI.

In this, one could add a second register in `ibex_core.sv` running on the ungated `clk_i` to flop the `fetch_enable_i` and then AND the output of this `clock_en`:
```
  always_ff @(posedge clk_i or negedge rst_ni) begin
    if (!rst_ni) begin
      fetch_enable_q <= 1'b0;
    end else if (fetch_enable_i) begin
      fetch_enable_q <= 1'b1;
    end
  end

  assign clock_en     = fetch_enable_q & (core_busy_q | debug_req_i | irq_pending | irq_nm_i);
```

bp-ouhaha, 2020-06-19:
that would be +/- equivalent to moving `ctrl_fsm_cs` from the gated to the ungated clock.

vogelpi, 2020-06-19:
Ah sorry, I meant to also remove `fetch_enable_i` from the controller.

I think it would be cleaner not to spread the ungated clock into more sub-modules (required to run `ctrl_fsm_cs` on the ungated clock) and at the same time remove the `fetch_enable_i` from the controller.

bp-ouhaha, 2020-06-19:
ok.

now I understand. Only make your suggested change in ibex_core.sv. That would work. The core would start a cycle later, not a big deal.

bp-ouhaha, 2020-06-19:
I'm testing it right now. single-cycle `fetch_enable_i` gets missed by ibex_controller.sv. I will reconnect ibex_controller/fetch_enable_i to fetch_enable_q.

also I found `core_busy_q` in ibex_core.sv is still clocked on the ungated-clock.

vogelpi, 2020-06-19:
You have to do three things:
1. Remove `fetch_enable_i` from the controller.
2. Set `ctrl_busy = 1'b0` inside the RESET state in the controller.
3. Add the `fetch_enable_q` flop as discussed above.

Btw: It is correct that `core_busy_q` is on the ungated glock.

bp-ouhaha, 2020-06-19:
@vogelpi 
I've done it. point 2 is not required (and make it fail actually to startup, ctrl_busy_o must be 1 to startup).
So 1 and 3 together work perfectly.

closed 2020-06-22 by vogelpi