lowRISC/ibex #964: Is parameter ICacheECC need to be passed by FuseSoC?

dawidzim, 2020-06-17:
Hey!
I have question about `ICacheECC `, is this parameter need to be passed by FuseSoC? I've problem with this, because when I generate files for Riviera and Xcelium by command `./vendor/lowrisc_ip/dvsim/dvsim.py dv/uvm/icache/dv/ibex_icache_sim_cfg.hjson -t [riviera/xcelium]` FuseSoC create: ` -pvalue+tb.ICacheECC=1` in file list.

rswarbrick, 2020-06-17:
The parameter is so that block-level testing of the ICache can choose whether to turn ECC on or not. At the moment, it's "hard-coded" at that point, but we'll eventually hook it up to Ibex's configuration machinery.

What's breaking here? Is it that the parameter value is getting included in a file that's supposed to be a list of SystemVerilog files? (If so, that sounds like it might be a problem with edalizer).

dawidzim, 2020-06-17:
When I run command: `./vendor/lowrisc_ip/dvsim/dvsim.py dv/uvm/icache/dv/ibex_icache_sim_cfg.hjson -t [riviera/xcelium]` FuseSoC generate files like for VCS, then in compilation we use `-f path/to/file.scr` and in this `file.scr` is `-pvalue+tb.ICacheECC=1`. 
So in Riviera compilation command going like this:
`vlog (...) -f path/to/file.scr` and we get error about unknown switch
In Riviera to pass value to this param we need to use `-g/tb.ICacheECC=1` in `vsim`, like is done in UVM core_ibex.
So to resolve this problem we need to change something in edalizer<?> or try to pass this argument like is done in UVM core_ibex.

imphil, 2020-06-17:
dvsim currently uses fusesoc/edalize to generate a VCS-specific file list (*.f) file, which happens to have parameters, defines, include paths, and files in it in VCS syntax. Dvsim then passes this file into its own flow, which supports multiple tools. 

This can work, but isn't guaranteed to work, as you found out: neither is the .f file produced by edalize a stable artifact of edalize, nor is the output format tool-agnostic enough to be used in all tools.

I don't think there's an easy solution at the moment.

- dvsim shouldn't rely on the *.f file; either it should use the stable *.eda.yml file, and generate a tool-specific file list itself, or use edalize completely to call the tool.
- I don't know if dvsim currently has a way to pass parameters/defines to tools in a tool-agnostic way. This would effectively require replicating the respective functionality from edalize.

I'll need to pass this one over to @sriyerg for more insight into workarounds and future plans.

sriyerg, 2020-06-17:
Please move the build / run time options from the fusesoc core file (`ibex_icache_sim.core`) to the dvsim config file (`ibex_icache_sim_cfg.hjson`) instead. We rely on fusesoc to only provide us a dependency resolved filelist devoid of any options. 

With dvsim, you can set elaboration time opts with tool-specific syntax through variable substitution within a 'build mode' a.k.a. a unique compile set. See example below:

```
  build_modes: [
    // Enable Icache ECC on all tests.
    {
      name: default
      en_build_modes: ["{tool}_icache_ecc"]
    }
    // Enable Icache ECC only on a subset of tests (not needed if enabling on all tests). 
    {
      name: icache_ecc
      en_build_modes: ["{tool}_icache_ecc"]
    }
    // Below options set Icache ECC param in a tool-specific way.
    {
      name: vcs_icache_ecc
      build_opts: ["-gv tb.ICacheECC=1"]
    }
    {
      name: riviera_icache_ecc
      build_opts: ["-g/tb.ICacheECC=1"] 
    }
   ...
  ]
  
  tests: [
    ...
    {
      name: ibex_icache_ecc_enabled_test
      // Indicate that this test uses a different compile-set than 'default'.
      build_mode: icache_ecc
    }
    ...
  ]
```

This is exactly how we use the tool-agnostic switches such as `--waves`, `--profile`, `--cov`, to enable waves, profiling and coverage respectively, which are highly tool specific. See `lowrisc/opentitan/hw/dv/data/common_modes.hjson`, which converts these agnostic 'build modes' into tool-specific ones (`lowrisc/opentitan/hw/dv/data/<tool>/<tool>.hjson`) using the variable substitution in the same way. 

Elaboration time settings such as this are problematic since there is no tool-standardized way of setting them. If it were a compile time setting (such as preprocessor macros with `+define+` or an additional include paths with `+incdir+`), this indirection would not have been needed, since they are standardized across all tools.

dawidzim, 2020-06-18:
Thanks! It works for me. Only one change, instead of:
```
    {
      name: riviera_icache_ecc
      build_opts: ["-g/tb.ICacheECC=1"] 
    }
```
we need to use:
```
    {
      name: riviera_icache_ecc
      run_opts: ["-g/tb.ICacheECC=1"] 
    }
```
Should I open pull request with this changes?

rswarbrick, 2020-06-18:
I'll do this today. Thanks for figuring out the right incantation with Riviera. I'll ping you on the PR.

closed 2020-06-22 by rswarbrick