lowRISC/ibex #971: Error running fusesoc on simple_system example

nagendragd, 2020-06-18:
I am new to this setup and got tools installed per the documentation (verilator, fusesoc, lowrisc-gcc)

When I run the fusesoc command specified to run the simple_system example, I get the error shown below:

Command that I am running: fusesoc --cores-root=. run --target=sim --setup --build lowrisc:ibex:ibex_simple_system --RV32M=1 --RV32E=0

Command returns: 
...
INFO: Preparing lowrisc:ibex:ibex_core:0.1
INFO: Preparing lowrisc:ibex:ibex_core_tracing:0.1
INFO: Preparing lowrisc:ibex:ibex_simple_system:0
INFO: Generating lowrisc:prim:prim_pkg-impl:0.1
Key 'cores' not found in GAPI structure. At least FuseSoC 1.11 is needed. Install a compatible version with 'pip3 install --user -r python-requirements.txt'.
ERROR: Setup failed : "python3 /home/UNT/ndg0068/rv_design/ibex/vendor/lowrisc_ip/prim/util/primgen.py /home/UNT/ndg0068/.cache/fusesoc/generated/lowrisc_prim_prim_pkg-impl_0.1/impl_input.yml" exited with an error code. See stderr for details.

I confirm that I have FuseSoc 1.11

fusesoc --version
1.11.1.dev1+g6f90890

imphil, 2020-06-18:
You need to install a patched version of fusesoc as the instructions say by running `pip3 install --user -r python-requirements.txt`. FuseSoC 1.11 was released two days ago, but still doesn't contain the necessary patches.

nagendragd, 2020-06-18:
Thank you - I just did that and I believe that step completed ok:
Installing collected packages: cssselect, cachetools, lxml, cssutils, premailer, bitstring, Pallets-Sphinx-Themes, sphinxcontrib-log-cabinet, sphinx-issues, smartypants, pdfrw, rst2pdf, edalize, fusesoc
  Running setup.py install for edalize ... done
  Running setup.py install for fusesoc ... done
Successfully installed Pallets-Sphinx-Themes-1.2.3 bitstring-3.1.7 cachetools-4.1.0 cssselect-1.1.0 cssutils-1.0.2 edalize-0.2.0 fusesoc-1.11.dev35+gd648377 lxml-4.5.1 pdfrw-0.4 premailer-3.7.0 rst2pdf-0.97 smartypants-2.0.1 sphinx-issues-1.2.0 sphinxcontrib-log-cabinet-1.0.1

However, when I ran the fusesoc command to build the simulator binary, I see the exact same error as I reported originally. I have provided the full log that I see when I run the fusesoc build:

ndg0068@alveo-cse:~/rv_design/ibex$ fusesoc --cores-root=. run --target=sim --setup --build lowrisc:ibex:ibex_simple_system --RV32M=1 --RV32E=0
WARNING: Unknown item version in section Root
WARNING: Unknown item formal in section Target
WARNING: Unknown item formal in section Target
INFO: Preparing lowrisc:dv_verilator:ibex_pcounts:0
INFO: Preparing lowrisc:dv_verilator:simutil_verilator:0
INFO: Preparing lowrisc:ibex:ibex_pkg:0.1
INFO: Preparing lowrisc:prim:assert:0.1
INFO: Preparing lowrisc:prim:primgen:0.1
INFO: Preparing lowrisc:prim:secded:0.1
INFO: Preparing lowrisc:tool:check_tool_requirements:0.1
INFO: Preparing lowrisc:dv_verilator:memutil_verilator:0
INFO: Preparing lowrisc:ibex:ibex_tracer:0.1
INFO: Preparing lowrisc:lint:common:0.1
INFO: Preparing lowrisc:prim:diff_decode:0
INFO: Preparing lowrisc:prim:prim_pkg:0.1
INFO: Preparing lowrisc:prim:clock_gating:0
INFO: Preparing lowrisc:prim:clock_mux2:0
INFO: Preparing lowrisc:prim:pad_wrapper:0
INFO: Preparing lowrisc:prim:ram_1p:0
INFO: Preparing lowrisc:prim:ram_2p:0
INFO: Preparing lowrisc:ibex:ibex_icache:0.1
INFO: Preparing lowrisc:ibex:sim_shared:0
INFO: Preparing lowrisc:prim:all:0.1
INFO: Preparing lowrisc:ibex:ibex_core:0.1
INFO: Preparing lowrisc:ibex:ibex_core_tracing:0.1
INFO: Preparing lowrisc:ibex:ibex_simple_system:0
INFO: Generating lowrisc:prim:prim_pkg-impl:0.1
Key 'cores' not found in GAPI structure. At least FuseSoC 1.11 is needed. Install a compatible version with 'pip3 install --user -r python-requirements.txt'.
ERROR: Setup failed : "python3 /home/UNT/ndg0068/rv_design/ibex/vendor/lowrisc_ip/prim/util/primgen.py /home/UNT/ndg0068/.cache/fusesoc/generated/lowrisc_prim_prim_pkg-impl_0.1/impl_input.yml" exited with an error code. See stderr for details.

imphil, 2020-06-18:
Can you do a `which fusesoc` and a `fusesoc --version` to double-check your fusesoc installation?

nagendragd, 2020-06-18:
Do the below look right?

ndg0068@alveo-cse:~/rv_design/ibex$ which fusesoc
/usr/local/bin/fusesoc
ndg0068@alveo-cse:~/rv_design/ibex$ fusesoc --version
1.11.1.dev1+g6f90890

imphil, 2020-06-18:
Looks like you have two fusesoc installations: one installed globally without the `--user` flag (that's the one in /usr/local/bin), and one that you just installed with the `--user` flag (that should be in ~/.local/bin).

Remove the globally installed one with `pip3 uninstall`, and then re-install to ensure you only have one version left.

If you go for the `--user` flag, make sure that ~/.local/bin is in your PATH.

nagendragd, 2020-06-18:
Ok thank you so much! That worked!!

nagendragd, 2020-06-18:
Looks like I didn't fix it. I deleted all installations and installed a system-wide install under /usr/local/bin/ and it reports version 1.11.0. However, when I run fusesoc, it is the same error. Even after running the requirements.txt, the same error repeats.
ndg0068@alveo-cse:~/rv_design/ibex$ which fusesoc
/usr/local/bin/fusesoc
ndg0068@alveo-cse:~/rv_design/ibex$ fusesoc --version
1.11.0
ndg0068@alveo-cse:~/rv_design/ibex$  fusesoc --cores-root=. run --target=sim --setup --build lowrisc:ibex:ibex_simple_system --RV32M=1 --RV32E=0

...
INFO: Preparing lowrisc:ibex:ibex_core:0.1
INFO: Preparing lowrisc:ibex:ibex_core_tracing:0.1
INFO: Preparing lowrisc:ibex:ibex_simple_system:0
INFO: Generating lowrisc:prim:prim_pkg-impl:0.1
Key 'cores' not found in GAPI structure. At least FuseSoC 1.11 is needed. Install a compatible version with 'pip3 install --user -r python-requirements.txt'.
ERROR: Setup failed : "python3 /home/UNT/ndg0068/rv_design/ibex/vendor/lowrisc_ip/prim/util/primgen.py /home/UNT/ndg0068/.cache/fusesoc/generated/lowrisc_prim_prim_pkg-impl_0.1/impl_input.yml" exited with an error code. See stderr for details.

imphil, 2020-06-18:
You're still on the wrong fusesoc version. How did you install the system wide version?

nagendragd, 2020-06-18:
This time, for system-wide installation, I ran:

sudo pip3 install --upgrade fusesoc