lowRISC/ibex #981: sv2v is not converting some systemVerilog files

nagendragd, 2020-06-22:
It appears that sv2v is unable to compile a few system Verilog sources to Verilog. When I ran sv2v (from within sys_yosys.sh script) with the verbose option, it reports:
sv2v: invalid for loop decl: // Trace: ../rtl/ibex_alu.sv:326:14
CallStack (from HasCallStack):
  error, called at src/Convert/ForDecl.hs:55:5 in main:Convert.ForDecl
sv2v: invalid for loop decl: // Trace: ../rtl/ibex_cs_registers.sv:875:14
CallStack (from HasCallStack):
  error, called at src/Convert/ForDecl.hs:55:5 in main:Convert.ForDecl
sv2v: invalid for loop decl: // Trace: ../rtl/ibex_icache.sv:363:14
CallStack (from HasCallStack):
  error, called at src/Convert/ForDecl.hs:55:5 in main:Convert.ForDecl
sv2v: invalid for loop decl: // Trace: ../rtl/ibex_pmp.sv:113:16
CallStack (from HasCallStack):
  error, called at src/Convert/ForDecl.hs:55:5 in main:Convert.ForDecl
sv2v: invalid for loop decl: // Trace: ../rtl/ibex_register_file_ff.sv:49:14
CallStack (from HasCallStack):
  error, called at src/Convert/ForDecl.hs:55:5 in main:Convert.ForDecl
sv2v: invalid for loop decl: // Trace: ../rtl/ibex_register_file_latch.sv:91:14
CallStack (from HasCallStack):
  error, called at src/Convert/ForDecl.hs:55:5 in main:Convert.ForDecl

Correspondingly, these files are not being translated to .v and the corresponding output files in the generated/ directory are 0 bytes. 

I tried both the downloadable executable from the sv2v webpage as well as installing a fresh one via git download and make. Both show the same results.

Any ideas how to solve this? (without modifying the source system verilog code..)

Thanks

Best Regards
Nagendra

imphil, 2020-06-22:
Thanks for your bug report. Please report this to sv2v upstream, there's probably nothing we can do here at Ibex.

nagendragd, 2020-06-22:
I am thinking it is something to do with my ibex setup if the same
documented flow has worked correctly for you all, no?

Thanks
Nagendra

nagendragd, 2020-06-23:
As you suggested, I filed an issue on sv2v and got it fixed. Thanks again!

closed 2020-06-23 by nagendragd

rswarbrick, 2020-06-23:
For anyone looking into this, the sv2v fix is at https://github.com/zachjs/sv2v/issues/98.