lowRISC/ibex #983: Debug single step test failure

tomroberts-lowrisc, 2020-06-23:
The debug single step test (small config, seed 1) is currently failing with the following message:
`DPC value [0x8000333e] is not the next instruction after ret_pc [0x8000b23a]`
As far as I can tell, the test is checking that DPC increments by 2 or 4 for every step taken. In this case, the instruction stepped over is a jump, and so the next PC does not equal the previous value - 2/4. The test is not expecting any jump/branch instructions to be generated, but there is one jump at the end (jump to test done) which it ends up stepping to, which causes the faliure.

udinator, 2020-06-23:
Hi @tomroberts-lowrisc , thanks for opening this issue. You are correct, the single step test is failing because of a new jump instruction right before the main program that got added in as a result of the modifications to program structure to support PMP.
I'm working on expanding this test to support branch and jump instructions, as that will help increase overall coverage.

udinator, 2020-06-30:
update: currently blocked on #987 - my updates to fix `debug_single_step_test` relies on the infrastructure introduced in this PR.

tomroberts-lowrisc, 2020-10-16:
I think this is now fixed - no longer seeing failures here so closing the issue

closed 2020-10-16 by tomroberts-lowrisc