lowRISC/ibex #985: [DV] Remove all copies of OpenTitan files except when vendored in

imphil, 2020-06-23:
We are vendoring in Ibex into OT, and we need to avoid files being duplicated, or core files in Ibex overriding ones in OT.

General rules:
- If code from OT is needed, it *must be* vendored in.
- If code is not vendored in, it *may not* use the same `name` in the core file, and ideally also not the same file names.

One example we know of right now is the mem_model at https://github.com/lowRISC/ibex/tree/master/dv/uvm/core_ibex/common/mem_model; there might be others.

This issue is to double-check all core files, and to ensure that the rules outlined hold.

rswarbrick, 2020-07-03:
I did a super-simple search for stuff duplicated between the two with these commands:

```sh
find opentitan -type f | grep -v '\.git' | grep -v 'vendor/' | xargs -n1 basename | sort -u >ot.txt
find ibex -type f | grep -v '\.git' | grep -v 'vendor/' | xargs -n1 basename | sort -u >ibex.txt
comm -12 ot.txt ibex.txt >both.txt
```

Obviously, this gives false positives (stuff like the azure pipeline), but it's a reasonable place to start. Grab the paths to these files, discarding Makefiles and readmes, in Ibex with:

```sh
xargs -n1 <both.txt find ibex -name | sort | grep -v vendor/ | grep -v Makefile | grep -v README.md >ibex-paths.txt
```

The resulting list ([ibex-paths.txt](https://github.com/lowRISC/ibex/files/4869366/ibex-paths.txt)) has 39 entries. I think the following items need looking at (in no particular order):

- `check_tool_requirements.core`, `util/check_tool_requirements.py`: We could vendor this in, I think.
- `dv/uvm/core_ibex/common/mem_model/*`: We could vendor this in.
- `dv/uvm/core_ibex/common/utils/clk_if.sv`, `dv/uvm/core_ibex/common/utils/dv_macros.svh`, `dv/uvm/core_ibex/common/utils/dv_utils_pkg.sv`: Manually copied to `dv/uvm/core_ibex` as a temporary hack. There's another copy properly vendored in. Long-term, the `core_ibex` tests should probably be ported to use fusesoc properly. Shorter term, we could probably just point them at the vendored code instead of the manually copied code.
- `dv/uvm/core_ibex/vcs.tcl`: This was manually copied from OT and the two versions have diverged. Getting this to work with the vendored version would take some work (that was done in OT in dvsim.py, which core_ibex doesn't use). Probably worth doing one day.
- `dv/uvm/data/*`: This was manually copied from OpenTitan into `dv/uvm/data`. Vendoring it in properly is tracked in issue #840, which needs [OpenTitan PR #2141](https://github.com/lowRISC/opentitan/pull/2141). *That* has been stuck in review since early May.
- `shared/rtl/fpga/xilinx/clkgen_xil7series.sv`: This file is used by several cores in OT and just one in Ibex. To vendor this properly, I think we'd need to give it its own core file. Probably not too hard.

The other items are:

- `azure-pipelines.yml`: These are just different, which is fine.
- `.clang-format`: These are different because of some special rules about autogenerated headers in OT. Probably fine to leave.
- `CONTRIBUTING.md`, `LICENSE`, `python-requirements.txt`: Top-level project stuff
- `*/tb.svg`: Testbench diagrams. These basenames aren't unique, so it's just a different testbench.
- `dv/uvm/core_ibex/cover.cfg`: Configuration of what coverage to collect. There's a certain amount of copying between the files, but they are distinct. Probably best to leave the manual copying for now, at least.
- `dv/uvm/icache/dv*`: These are testbench-specific files (which don't have unique basenames).
- `*/coremark/*`: We've done the vendoring properly here already, I think.
- `syn/lec_sv2v.do`, `syn/syn_yosys.sh`: These are different between the projects, if related, and have explicit notes saying so.
- `tool_requirements.py`: Different between the projects, so we should keep them separate.

I propose working through the first list above in the following order:

1. The code in `dv/uvm/core_ibex/common/utils`. Doing the hacky fix (changing to a different set of hardcoded paths) should be easy enough. Let's just do it.
1. The memory model. The OT and Ibex versions have diverged, but I think this is just new functionality in the OT version. We should just vendor it in to Ibex.
1. `check_tool_requirements.*`
1. `dv/uvm/data/*`: High priority, but stuck in review for what feels like forever.
1. `vcs.tcl`. Not massively high priority, but probably worth doing eventually. If nothing else, I think the code that's currently there won't work to get Verdi for anyone outside of Google. (I fixed the equivalent bug in the OpenTitan code a while ago).
1. `shared/rtl/fpga/xilinx/clkgen_xil7series.sv`: Pretty low priority, I would think?

@imphil: Does that list and/or ordering sound sensible to you?

imphil, 2020-07-03:
The list looks good!

rswarbrick, 2020-07-21:
I spoke too soon when I claimed that (1.) should be "easy enough" above. It turns out that the correctly vendored files depend on a top_pkg core file which we (wrongly) have a duplicate of in the icache test directory. PR #1042 tidies this up, breaking the tight coupling with OpenTitan by defining an Ibex flavour of the core.

rswarbrick, 2020-07-23:
The second check-box on the list above should be addressed by #1047.