lowRISC/ibex #1006: Yosys syn flow: Use fusesoc

imphil, 2020-07-03:
The Yosys flow in `syn` currently ingests source files from the rtl directory and other places, feeds them through sv2v, and then passes them through yosys.

Since https://github.com/lowRISC/ibex/pull/994 we have a nicer way of calling sv2v through fusesoc; and a yosys-without-sv2v flow might not be too far away either. 

Depending on how things evolve, we have two options to use fusesoc for the syn flow, and make use of its ability to generate target-dependent primitives, like the clock gating ones.

imphil, 2020-07-16:
Related to #501 for the clock gating primitive.

towoe, 2020-08-18:
> and a yosys-without-sv2v flow might not be too far away either.
> 
@imphil do you mean by that a sufficient SystemVerilog support by yosys?

imphil, 2020-08-19:
Yes. Antmicro is using Ibex as test bed for Yosys improvements, and all necessary patches are available in https://github.com/antmicro/ibex-yosys-build. These patches are now making slowly their way upstream.

imphil, 2020-08-20:
See https://github.com/lowRISC/ibex/issues/60 for details of the remaining issues we know about.