lowRISC/ibex #1018: Illegal Instruction Exception not Raised - Shift Instructions

shetalani, 2020-07-07:
**RISC-V Specification:**

- _"Chapter 2.4 of Unprivileged ISA"_

**Issue Description:**

An illegal instruction updates the register file and the associated csrs and executes normally without raising an illegal instruction exception.

**Example:**

The instruction 32'h02005913 is decoded at time point t##0 with no illegal instruction being flagged, as illegal_insn_did is de-asserted.

[image]

---

Product: OneSpin 360 DV-Verify
App: RVV
Tool's version: 2020.1.3

imphil, 2020-07-07:
Thanks @shetalani for your bug report.

Just to confirm we're speaking about the same issue. The instruction you're seeing would be decoded as:

```
02005913          srli          s2,zero,32
```

And the issue being that the immediate should only be 5 bit wide, and 32 isn't a valid immediate ("the shift amount is encoded in the lower 5 bits of the I-immediate field." as the spec says)

shetalani, 2020-07-07:
Yes, on the other hand, the spec states that the upper 7 bits should be zero for such an instruction to be decoded as SRLI.

vogelpi, 2020-07-16:
Hi @shetalani ,
I have created a PR to solve the issue [here](#1036).

vogelpi, 2020-07-16:
The actual problem was that the decoder did not check bits 26 and 25 for slli, srli and srai instructions.

closed 2020-07-17 by vogelpi

imphil, 2020-07-17:
@shetalani can you re-run the tool to check if the issue is now fully resolved? The latest `master` version of Ibex should have the required fix.

shetalani, 2020-10-13:
The issue is fixed, cheers!