lowRISC/ibex #1022: How to use malloc() for IBEX simple_system

mehmood-007, 2020-07-07:
Hi,

Is there any possibility to allocate memory malloc() with the IBEX simple_system environment using the newlib RISCV gnu toolchain? As that function make a system call and simple_system use a bare-metal environment. Is there anything we need to define in the linker script for heap allocation. Furthermore, if anyone suggests a good alternative for using file system based system calls like fread, fseek , etc it will be really helpful. As I have to run the benchmarks (Mibench) on IBEX, and facing difficulties with these system calls, as most of the algorithms require these system calls.

imphil, 2020-07-07:
Thanks @mehmood-007 for your question. 

Ibex and the simple system are meant to demonstrate the Ibex, that is, the CPU core hardware. Your question is rather independent of Ibex, and more about software for RISC-V in general. Unfortunately, this issue tracker is the wrong place to ask such questions (and we're the wrong group of people to ask).

Regarding benchmarks: Generally, all benchmarks which require file I/O are hard or impossible to port to deeply embedded systems like the ones Ibex is typically used for. If you're looking for good benchmarks in that domain, I can recommend CoreMark (for which we ship a configuration in this repository), and Embench.

mehmood-007, 2020-07-07:
Thanks @imphil, I will use CoreMark and Embench for our research.

closed 2020-07-07 by mehmood-007