lowRISC/ibex #1030: g++: error: unrecognized command line option ‘-faligned-new’

islengxw, 2020-07-10:
Hi
I am try to run simple system hello test.
when i use below command to build the simulation
"fusesoc --cores-root=. run --target=sim --setup --build lowrisc:ibex:ibex_simple_system --RV32M=1 --RV32E=0"
But i got the error message, my g++ version is 4.8.
Any suggestion will be welcome.

Thank you

```
g++  -I.  -MMD -I/tools/verilator/4.028/share/verilator/include -I/tools/verilator/4.028/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -faligned-new -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -I../src/lowrisc_dv_verilator_ibex_pcounts_0/cpp -I../src/lowrisc_dv_verilator_simutil_verilator_0/cpp -I../src/lowrisc_prim_assert_0.1/rtl -I../src/lowrisc_prim_util_memload_0/rtl -I../src/lowrisc_dv_verilator_memutil_verilator_0/cpp -std=c++11 -Wall -DVM_TRACE_FMT_FST -DTOPLEVEL_NAME=ibex_simple_system -g   -c -o ibex_pcounts.o ../src/lowrisc_dv_verilator_ibex_pcounts_0/cpp/ibex_pcounts.cc
g++  -I.  -MMD -I/tools/verilator/4.028/share/verilator/include -I/tools/verilator/4.028/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -faligned-new -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -I../src/lowrisc_dv_verilator_ibex_pcounts_0/cpp -I../src/lowrisc_dv_verilator_simutil_verilator_0/cpp -I../src/lowrisc_prim_assert_0.1/rtl -I../src/lowrisc_prim_util_memload_0/rtl -I../src/lowrisc_dv_verilator_memutil_verilator_0/cpp -std=c++11 -Wall -DVM_TRACE_FMT_FST -DTOPLEVEL_NAME=ibex_simple_system -g   -c -o verilator_memutil.o ../src/lowrisc_dv_verilator_memutil_verilator_0/cpp/verilator_memutil.cc
g++: error: unrecognized command line option ‘-faligned-new’
Vibex_simple_system.mk:76: recipe for target 'verilator_memutil.o' failed
make[1]: *** [verilator_memutil.o] Error 1
make[1]: *** Waiting for unfinished jobs....
g++: error: unrecognized command line option ‘-faligned-new’
Vibex_simple_system.mk:74: recipe for target 'ibex_pcounts.o' failed
```

rswarbrick, 2020-07-10:
The problem here is that Verilator is emitting the `-faligned-new` flag and your GCC is too old. I don't think that Verilator lists a required GCC version, but Verilator 4.028 is very recent and GCC 4.8 is ancient (released in 2013!). I think you just need to install a more recent GCC.

Note that, even if you're on an old system, you can often install newer GCC versions to somewhere nonstandard (/opt, for example). Then you just need to prepend that directory to your path or alternatively set the `CC` and `CXX` environment variables to tell Verilator which compiler to use.

imphil, 2020-07-10:
GCC 4.8 should work fine with Verilator, as long as you disable C++11 when configuring Verilator at build time.

If I remember correctly, Verilator only adds `-faligned-new` to its flags if it was deemed safe so at configure time. So please double-check the way you compile/install Verilator, and file a bug at https://github.com/verilator/verilator to get more help otherwise. There's unfortunately nothing we can do here on the Ibex side.

closed 2020-07-10 by imphil