lowRISC/ibex #1033: `Symbiyosis` default tool in `ibex_icache_fpv.core`

msfschaffner, 2020-07-14:
I am getting a bunch of the following warnings after the newest vendor-in of Ibex in OpenTitan:

`WARNING: Unknown item symbiyosys in section Tools`

The cause seems to be the tool assignment in:
https://github.com/lowRISC/ibex/blob/6b9165fa66b49534226acfcb739f2c252be4853c/formal/icache/ibex_icache_fpv.core#L62-L73

Does that rely on a custom / unreleased version of Edalize?

If so, can we upstream that or fix that otherwise?

(I waived the warning in the meantime in order to get style-lint clean.)

rswarbrick, 2020-07-15:
Yes, it's in a currently unreleased version of Edalize. The PR is here: https://github.com/olofk/edalize/pull/173 (and apparently needs rebasing: I'll do that tomorrow).

I'd hope to get it merged soonish (@imphil: does that sound plausible to you?). If this is a big problem, we could pull the patches into the "ot" branch of Edalize, but I would rather just get it into upstream first (rather than having to fiddle with possible changes as the PR gets merged).

rswarbrick, 2021-01-25:
The upstream patch was merged into mainline in August, so everything should quiet again now.

@msfschaffner: Do you need to remove a waiver from somewhere in OpenTitan?

rswarbrick, 2021-02-01:
I agree there's something to do on the OT side, but don't think this needs tracking in Ibex, so re-closing here.

closed 2021-02-01 by rswarbrick

msfschaffner, 2021-02-01:
Woops, sorry @rswarbrick. I did not realize that this was not an OT issue.