lowRISC/ibex #1038: Sporadic failures in `riscv_illegal_instr_test`

GregAC, 2020-07-17:
@vogelpi and I have both seen some failures in the `riscv_illegal_instr_test`. I did look into some and I believe they're happening because when running with Spike as a simulator the 'F' extension and potential others are turned on. So illegal instruction for Ibex get executed on Spike and a trace mis-match results.

We should look into the failures in more detail and if my diagnosis is correct ensure Spike runs with the correct ISA.

vogelpi, 2020-07-17:
Thanks Greg for opening this issue. I have tracked down the bug. You can use the following command to reproduce the issue:
```
make TEST=riscv_illegal_instr_test ITERATIONS=1 SEED=29633 SIMULATOR=vcs ISS=spike
```
This should give the following regression log:
```
Test binary: out/seed-29633/instr_gen/asm_tests/riscv_illegal_instr_test.0.o
ibex uvm log : out/seed-29633/rtl_sim/riscv_illegal_instr_test.0/sim.log
riscv_illegal_instr_test : [PASSED]

ibex : out/seed-29633/rtl_sim/riscv_illegal_instr_test.0/trace_core_00000000.csv
spike : out/seed-29633/instr_gen/spike_sim/riscv_illegal_instr_test.0.csv
Mismatch[1]:
ibex[4305] : pc[8000491c] csrrc	x7,mscratch,x10: t2:0000001e 
spike[4305] : pc[ffffffff80001380] addi    a3, a3, -124: a3:8001efe8 
Mismatch[2]:
ibex[4306] : pc[80004920] xori	x10,x19,747: a0:000002e9 
spike[4306] : pc[ffffffff800013f0] csrr    a2, mepc: a2:80004918 
Mismatch[3]:
ibex[4307] : pc[80004924] and	x5,x5,x11: t0:00000000 
spike[4307] : pc[ffffffff800013f4] csrr    a2, mcause: a2:00000002 
Mismatch[4]:
ibex[4308] : pc[80004928] xori	x16,x5,274: a6:00000112 
spike[4308] : pc[ffffffff800013f8] c.li    t6, 3: t6:00000003 
Mismatch[5]:
ibex[4309] : pc[8000492c] csrrwi	x27,mscratch,24: s11:00000000 
spike[4309] : pc[ffffffff80001402] c.li    t6, 8: t6:00000008 
24 instructions left in trace spike
[FAILED]: 2942 matched, 16896 mismatch

0 PASSED, 1 FAILED
```

Looking at the the Spike log we can see that an illegal instruction exception is thrown by Spike when executing the following instruction:
```
core   0: 0xffffffff80004918 (0x7c09a1f3) csrrs   gp, unknown_7c0, s3
core   0: exception trap_illegal_instruction, epc 0xffffffff80004918
```

The reason is that the referenced CSR at address 0x7C0 is a custom one. The spec mandates [Sec 2.1 of the RISC-V Priv. Spec.]:
"Attempts to access a non-existent CSR raise an illegal instruction exception."

However, this CSR is implemented in Ibex (together with others), see `ibex_pgk.sv`:
```
  CSR_CPUCTRL        = 12'h7C0,
  CSR_SECURESEED     = 12'h7C1
``` 
Ibex does not throw an exception. That's why we see a difference between Spike and the RTL simulation.

I think there should be a way to tell the simulator which CSRs are implemented. Probably inside `dv/uvm/core_ibex/riscv_dv_extension/csr_description.yaml` but I don't know the syntax. Will have to check on Monday.

udinator, 2020-07-20:
Hi @vogelpi , @GregAC , thanks for opening this issue!
Pirmin is correct that the failures are due to Ibex supporting custom CSRs that riscv-dv is not aware of; I'm working on implementing support for this on the riscv-dv end.

vogelpi, 2020-07-20:
Hey @udinator thanks for looking into this on the riscv-dv end. I did a draft PR to include the description of the custom CSRs. I guess you will be able to re-use this.

udinator, 2020-07-23:
Update: #1049 and #1050 fix the illegal_instr_test failures that are due to accesses to `cpuctrl` and `secureseed`.

There still remains one class of failures in this test which is caused by riscv-dv generating valid B-extension instructions as illegal instructions as it doesn't know any better - this will be likely much more involved to fix.

udinator, 2020-08-10:
Update: the issue has been fixed with a recent vendor PRA from riscv-dv, so I will close this issue as the illegal instruction tests are all passing successfully now.

closed 2020-08-10 by udinator

vogelpi, 2020-08-10:
Thanks @udinator for solving this issue :+1:

d-thakkar, 2021-02-23:
Hi, was this issue resolved for all the tests? I have the csr_description.yaml file with updates about 0x7c0 and 0x7c1 custom CSRs.

I ran the test with following arguments,
make TEST=riscv_arithmetic_basic_test ITERATIONS=1 VERBOSE=1 WAVES=1 SIMULATOR=vcs ISS=spike

and I see spike log files with similar error message, related to trap_illegal_transaction.

This is what regr.log file prints,

Mismatch[1]:
ibex[109] : pc[80000256] lui	x17,0x2: a7:00002000 
spike[109] : pc[ffffffff80001580] addi    a5, a5, -124: a5:ffffff84 
Mismatch[2]:
ibex[110] : pc[8000025a] addi	x17,x17,-2048: a7:00001800 
spike[110] : pc[ffffffff80001580] addi    a5, a5, -124: a5:ffffff08 
Mismatch[3]:
ibex[111] : pc[80000262] c.li	x17,0: a7:00000000 
spike[111] : pc[ffffffff800015f0] csrr    a7, mepc: a7:800015ec 
Mismatch[4]:
ibex[112] : pc[80000274] lui	x1,0xf2bae: ra:f2bae000 
spike[112] : pc[ffffffff800015f4] csrr    a7, mcause: a7:00000007 
Mismatch[5]:
ibex[113] : pc[80000278] addi	x1,x1,-1599: ra:f2bad9c1 
spike[113] : pc[ffffffff800015f8] c.li    a1, 3: a1:00000003 
223716 instructions left in trace spike
[FAILED]: 118 matched, 231731 mismatch

Spike log file
core   0: 0xffffffff80000252 (0x7c00d073) csrwi   unknown_7c0, 1
core   0: exception trap_illegal_instruction, epc 0xffffffff80000252
core   0:           tval 0x000000007c00d073
core   0: >>>>  mtvec_handler

Looks like the problem is similar to what is mentioned in this issue, but not sure the reason why? Can anyone help me explain?

Also, is there a way to disable sending CSR instructions to custom CSRs 0x7c0 and 0x7c1, if so how can we do that ? 

For my application I will not be implementing custom CSRs anyways.

GregAC, 2021-02-23:
Are you using the lowRISC version of Spike(https://github.com/lowRISC/riscv-isa-sim/tree/ibex)? It looks like riscv_arithmetic_basic_test is producing writes to the `CSR_CPUCTRL` custom CSR (used to enable/disable ICache) which needs the custom Spike version to work.

Are you using the ICache? It should be possible to disable writes to this CSR for testing configurations that don't use the ICache though this may need to be done manually. Commenting out the line `instr.push_back("csrwi 0x7c0, 1");` from `dv/uvm/core_ibex/riscv_dv_extension/ibex_asm_program_gen.sv` should do this, @udinator may know more.

udinator, 2021-02-23:
Hi @d-thakkar, like @GregAC mentioned it looks like you may be using an "incorrect" version of Spike (it's strongly recommended to use the lowRISC version of Spike with Ibex to avoid unexpected failures like the one you are seeing).

I believe Greg's workaround solution would work as well, feel free to try it out if you are unable to use the recommended Spike.

d-thakkar, 2021-02-24:
@GregAC and @udinator Thanks for helping. I was not using the lowrisc/custom spike version. I was using the generic one. For my application I won't be using ICache so yes disabling the csr writes to custom CSR helped me to move forward.