lowRISC/ibex #1041: ASIC-proven, base clock frequency and technology details

machkar, 2020-07-20:
Hello All,

This is not really an issue but could not find this information anywhere.
Has any version of this project been silicon-proven, if so at what technology (nm) and is there a possible operable ASIC clock frequency range.

Thanks a lot.

GregAC, 2020-07-20:
I'm afraid we don't have guidance on achievable frequency to hand. The OpenTitan project does have some limited information from a nightly implementation flow: https://reports.opentitan.org/hw/top_earlgrey/syn/2020.07.16_12.16.19/results.html. It includes an Ibex core amongst other pieces of IP. Ibex is on the running on MAIN_CLK which is reported to have a period of 8ns without any timing issues. Sadly I don't think the technology node that is running on is publicly available.

Zero-Riscy, which is what Ibex used to be called has seen several tape-outs under the PULP project you can see details of chips they have built here: https://pulp-platform.org/implementation.html e.g. http://asic.ethz.ch/2017/Mr.Wolf.html contains a Zero-Riscy core.

I would caution that since lowRISC took on development of Zero-Riscy (where it was renamed Ibex) significant development has occurred so what is currently in the Ibex master branch will have differences from what the PULP project have taped out.

imphil, 2020-10-23:
The Ibex README has since updated with some timing and area data. Please note that you'll still need to run own trials in your synthesis flow to get trustworthy numbers for a particular setup.

closed 2020-10-23 by imphil