lowRISC/ibex #1044: PMP TOR writes and behavior

sorear, 2020-07-22:
1. There appears to be a functional difference between [spike](https://github.com/riscv/riscv-isa-sim/blob/67b7edd0270d5d074136eb45474477d15aeee725/riscv/processor.cc#L809) and [ibex](https://github.com/lowRISC/ibex/blob/d7c13aa9045838e7b9751e87b6b4630e7893db39/rtl/ibex_cs_registers.sv#L828) in the interaction between L and A=TOR; riscv/riscv-isa-manual#545 endorsed the spike interpretation.

2. Should https://github.com/lowRISC/ibex/blob/master/rtl/ibex_pmp.sv#L92-L93 be generating both ends of the TOR range from the same pmpaddr?

tomroberts-lowrisc, 2020-07-22:
1. Do we differ from spike here? Locking of the address registers is implemented [here](https://github.com/lowRISC/ibex/blob/d7c13aa9045838e7b9751e87b6b4630e7893db39/rtl/ibex_cs_registers.sv#L827). I think we verified the Ibex implementation against spike - @udinator will be able to confirm

2. The start address of the TOR range comes from `region_start_address` which will be the previous PMP region address - see [here](https://github.com/lowRISC/ibex/blob/d7c13aa9045838e7b9751e87b6b4630e7893db39/rtl/ibex_pmp.sv#L53)

GregAC, 2020-07-22:
On 1. I think we'll disallow writes to `pmp_addr[i]` if entry `i+1` has A=TOR regardless of the lock bit on entry `i+1` where the spec update (which matches current Spike behavior) indicates we should allow those unless entry `i+1` has both A=TOR and the lock bit set?

tomroberts-lowrisc, 2020-07-22:
Yep - just re-read it myself - will fix to match the new spec for 1.

rswarbrick, 2020-07-22:
Does this just show that we haven't tried these tests against a recent Spike, or is it a proper test hole?

GregAC, 2020-07-22:
We should also check what the OVPsim behavior is here. Given our behavior was spec compliant until https://github.com/riscv/riscv-isa-manual/pull/545 went in 7 hours ago OVPsim may also have similar behavior and not match Spike.

udinator, 2020-07-22:
@tomroberts-lowrisc - yes, we verified the Ibex implementation against both Spike and OVPsim, not taking into account this recent spec change.
@rswarbrick to your point - yes, it means that we haven't tried these tests against recent versions of Spike and OVPsim, so I'll need to update both and verify the Ibex implementation in this new context.
It might be worth adding an explicit test for this feature in the spec just to make sure as well.

sorear, 2020-07-25:
@udinator the Spike behavior has been [the same since it went in in 2018] though?

udinator, 2020-07-29:
@sorear: good point - after investigating more deeply this indeed seems to be a small test hole, the existing suite of PMP tests exercises repeated writes to the `pmpcfg` CSRs (which work as intended), just not to the `pmpaddr` CSRs.

@GregAC @tomroberts-lowrisc I've been able to reproduce this issue with some updates to the current PMP test suite, and I've opened #1054 to fix the RTL issue.

imphil, 2020-10-23:
Fixed in #1054

closed 2020-10-23 by imphil