lowRISC/ibex #1051: [uvm] 'custom_csr' has not been declared yet

GregAC, 2020-07-24:
I just tried running the DV on latest master and received the following error:

```
Error-[IND] Identifier not declared
/home/greg/work/ibex_area/ibex/vendor/google_riscv-dv/src/riscv_illegal_instr.sv, 1
  Identifier 'custom_csr' has not been declared yet. If this error is not
  expected, please check if you have set `default_nettype to none.
```

It comes from the latest vendoring in of riscv-dv with the new custom_csr stuff. Does something need configuring in the Ibex repository as well for this to work?

udinator, 2020-07-24:
Yes, this relies on #1050 to compile and run properly.

GregAC, 2020-07-24:
Great thanks for the quick resolution!

closed 2020-07-27 by udinator