lowRISC/ibex #1071: Cannot find assembly test

aliess, 2020-08-12:
When I am trying to run UVM tests, using the below command to generate `riscv_jump_stress_test` for example:

`make SIMULATOR=questa TEST=riscv_jump_stress_test ISS=spike ISA=rv32imc`

will result in the below error log: 

```
Wed, 12 Aug 2020 18:33:46 INFO     Creating output directory: out/seed-25749/instr_gen
Wed, 12 Aug 2020 18:33:46 INFO     Processing regression test list : riscv_dv_extension/testlist.yaml, test: riscv_jump_stress_test
Wed, 12 Aug 2020 18:33:46 INFO     Found matched tests: riscv_jump_stress_test, iterations:1
Wed, 12 Aug 2020 18:33:46 INFO     Processing simulator setup file : /home/ali/Documents/ibex/vendor/google_riscv-dv/yaml/simulator.yaml
Wed, 12 Aug 2020 18:33:46 INFO     Found matching simulator: questa
Wed, 12 Aug 2020 18:33:46 INFO     Building RISC-V instruction generator
Wed, 12 Aug 2020 18:33:51 INFO     Running RISC-V instruction generator
Wed, 12 Aug 2020 18:33:51 INFO     Generating 1 riscv_jump_stress_test
Wed, 12 Aug 2020 18:33:51 INFO     Running riscv_jump_stress_test with 1 batches
Wed, 12 Aug 2020 18:33:51 INFO     Running riscv_jump_stress_test, batch 1/1, test_cnt:1
Wed, 12 Aug 2020 18:33:57 INFO     Creating output directory: out/seed-25749/instr_gen
Wed, 12 Aug 2020 18:33:57 INFO     Processing regression test list : riscv_dv_extension/testlist.yaml, test: riscv_jump_stress_test
Wed, 12 Aug 2020 18:33:57 INFO     Found matched tests: riscv_jump_stress_test, iterations:1
Wed, 12 Aug 2020 18:33:57 ERROR    Cannot find assembly test: out/seed-25749/instr_gen/asm_tests/riscv_jump_stress_test_0.S

make: *** [Makefile:273: out/seed-25749/.metadata/instr_gen.compile.stamp] Error 1
```

The assembly file in indeed not generated and not found in the `out` directory, and I can't seem to find any error python is throwing in order to debug the issue.

This doesn't happen when testing `riscv_arithmetic_basic_test` using the below command: 

`make SIMULATOR=questa TEST=riscv_arithmetic_basic_test ISS=spike ISA=rv32imc`

This would run normally and finish successfully, however every other test other than the `riscv_arithmetic_basic_test` will fail as above.

rswarbrick, 2020-08-12:
This is a little tricky to debug because the Makefile doesn't spit out commands as it goes (not my preferred style!). Can you run an example again with something like this?
```
make --trace SIMULATOR=questa TEST=riscv_jump_stress_test ISS=spike ISA=rv32imc SEED=1 ITERATIONS=1
```
That should show exactly which command is failing.

Also, could you look in your build directory? I end up with `out/seed-1/instr_gen/asm_tests/riscv_jump_stress_test_0.S`: I assume that file is missing? If so, it looks like something isn't running correctly with the instruction generator. You can run that on its own with the `gen` target (add "gen" to the end of the command line above). As a sanity check, can you search for `.S` files in the build tree with something like `find -name '*.S' out/seed-1`?

rswarbrick, 2020-08-12:
As a note: We use VCS, rather than Questa, at lowRISC. It looks like there should be support for Questa, but it's possible that something has broken with that without us noticing. If so, thanks for the bug report(!) but we might have to do a bit of back-and-forth to work out what's not working.

aliess, 2020-08-12:
using the command you provided:

`make --trace SIMULATOR=questa TEST=riscv_jump_stress_test ISS=spike ISA=rv32imc SEED=1 ITERATIONS=1`

produces the below log: 
```
Makefile:125: target 'out' does not exist
mkdir -p out
Makefile:142: update target 'out/.sim-cfg.mk' due to: FORCE
./sim_makefrag_gen.py questa experimental-maxperf-pmp-bmfull /home/ali/Documents/ibex out/.sim-cfg.mk
Makefile:125: target 'out/seed-1/.metadata' does not exist
mkdir -p out/seed-1/.metadata
Makefile:247: update target 'out/seed-1/.metadata/instr_gen.gen.stamp' due to: FORCE /home/ali/Documents/ibex/vendor/google_riscv-dv/README.md /home/ali/Documents/ibex/vendor/google_riscv-dv/test/riscv_instr_base_test.sv /home/ali/Documents/ibex/vendor/google_riscv-dv/test/riscv_instr_cov_test.sv /home/ali/Documents/ibex/vendor/google_riscv-dv/test/riscv_instr_test.sv /home/ali/Documents/ibex/vendor/google_riscv-dv/test/riscv_instr_gen_tb_top.sv /home/ali/Documents/ibex/vendor/google_riscv-dv/test/riscv_instr_test_lib.sv /home/ali/Documents/ibex/vendor/google_riscv-dv/test/riscv_instr_test_pkg.sv /home/ali/Documents/ibex/vendor/google_riscv-dv/setup.cfg /home/ali/Documents/ibex/vendor/google_riscv-dv/design.bin /home/ali/Documents/ibex/vendor/google_riscv-dv/scripts/link.ld /home/ali/Documents/ibex/vendor/google_riscv-dv/scripts/ovpsim_log_to_trace_csv.py /home/ali/Documents/ibex/vendor/google_riscv-dv/scripts/sail_log_to_trace_csv.py /home/ali/Documents/ibex/vendor/google_riscv-dv/scripts/deprecated/ovpsim_log_to_trace_csv.py /home/ali/Documents/ibex/vendor/google_riscv-dv/scripts/deprecated/spike_log_to_trace_csv.py /home/ali/Documents/ibex/vendor/google_riscv-dv/scripts/deprecated/riscv_trace_csv.py /home/ali/Documents/ibex/vendor/google_riscv-dv/scripts/deprecated/instr_trace_compare.py /home/ali/Documents/ibex/vendor/google_riscv-dv/scripts/gen_csr_test.py /home/ali/Documents/ibex/vendor/google_riscv-dv/scripts/__pycache__/ovpsim_log_to_trace_csv.cpython-38.pyc /home/ali/Documents/ibex/vendor/google_riscv-dv/scripts/__pycache__/riscv_trace_csv.cpython-38.pyc /home/ali/Documents/ibex/vendor/google_riscv-dv/scripts/__pycache__/instr_trace_compare.cpython-38.pyc /home/ali/Documents/ibex/vendor/google_riscv-dv/scripts/__pycache__/whisper_log_trace_csv.cpython-38.pyc /home/ali/Documents/ibex/vendor/google_riscv-dv/scripts/__pycache__/sail_log_to_trace_csv.cpython-38.pyc /home/ali/Documents/ibex/vendor/google_riscv-dv/scripts/__pycache__/spike_log_to_trace_csv.cpython-38.pyc /home/ali/Documents/ibex/vendor/google_riscv-dv/scripts/__pycache__/lib.cpython-38.pyc /home/ali/Documents/ibex/vendor/google_riscv-dv/scripts/spike_log_to_trace_csv.py /home/ali/Documents/ibex/vendor/google_riscv-dv/scripts/riscv_trace_csv.py /home/ali/Documents/ibex/vendor/google_riscv-dv/scripts/whisper_log_trace_csv.py /home/ali/Documents/ibex/vendor/google_riscv-dv/scripts/instr_trace_compare.py /home/ali/Documents/ibex/vendor/google_riscv-dv/scripts/lib.py /home/ali/Documents/ibex/vendor/google_riscv-dv/run.py /home/ali/Documents/ibex/vendor/google_riscv-dv/target/rv64gcv/riscv_core_setting.sv /home/ali/Documents/ibex/vendor/google_riscv-dv/target/rv64gcv/testlist.yaml /home/ali/Documents/ibex/vendor/google_riscv-dv/target/rv64gcv/riscvOVPsim.ic /home/ali/Documents/ibex/vendor/google_riscv-dv/target/rv32i/riscv_core_setting.sv /home/ali/Documents/ibex/vendor/google_riscv-dv/target/rv32i/testlist.yaml /home/ali/Documents/ibex/vendor/google_riscv-dv/target/rv32i/riscvOVPsim.ic /home/ali/Documents/ibex/vendor/google_riscv-dv/target/multi_harts/riscv_core_setting.sv /home/ali/Documents/ibex/vendor/google_riscv-dv/target/multi_harts/testlist.yaml /home/ali/Documents/ibex/vendor/google_riscv-dv/target/multi_harts/riscvOVPsim.ic /home/ali/Documents/ibex/vendor/google_riscv-dv/target/rv64imcb/riscv_core_setting.sv /home/ali/Documents/ibex/vendor/google_riscv-dv/target/rv64imcb/testlist.yaml /home/ali/Documents/ibex/vendor/google_riscv-dv/target/rv64imcb/riscvOVPsim.ic /home/ali/Documents/ibex/vendor/google_riscv-dv/target/rv64gc/riscv_core_setting.sv /home/ali/Documents/ibex/vendor/google_riscv-dv/target/rv64gc/testlist.yaml /home/ali/Documents/ibex/vendor/google_riscv-dv/target/rv64gc/riscvOVPsim.ic /home/ali/Documents/ibex/vendor/google_riscv-dv/target/rv64imc/riscv_core_setting.sv /home/ali/Documents/ibex/vendor/google_riscv-dv/target/rv64imc/testlist.yaml /home/ali/Documents/ibex/vendor/google_riscv-dv/target/rv64imc/riscvOVPsim.ic /home/ali/Documents/ibex/vendor/google_riscv-dv/target/rv32imc/riscv_core_setting.sv /home/ali/Documents/ibex/vendor/google_riscv-dv/target/rv32imc/testlist.yaml /home/ali/Documents/ibex/vendor/google_riscv-dv/target/rv32imc/riscvOVPsim.ic /home/ali/Documents/ibex/vendor/google_riscv-dv/target/rv32imcb/riscv_core_setting.sv /home/ali/Documents/ibex/vendor/google_riscv-dv/target/rv32imcb/testlist.yaml /home/ali/Documents/ibex/vendor/google_riscv-dv/target/rv32imcb/riscvOVPsim.ic /home/ali/Documents/ibex/vendor/google_riscv-dv/target/ml/riscv_core_setting.sv /home/ali/Documents/ibex/vendor/google_riscv-dv/target/ml/testlist.yaml /home/ali/Documents/ibex/vendor/google_riscv-dv/target/ml/riscvOVPsim.ic /home/ali/Documents/ibex/vendor/google_riscv-dv/MANIFEST.in /home/ali/Documents/ibex/vendor/google_riscv-dv/cov.py /home/ali/Documents/ibex/vendor/google_riscv-dv/.flake8 /home/ali/Documents/ibex/vendor/google_riscv-dv/pygen/pygen_src/riscv_asm_program_gen.py /home/ali/Documents/ibex/vendor/google_riscv-dv/pygen/pygen_src/test/out/.gitkeep /h [... cut ...]
Wed, 12 Aug 2020 19:14:35 INFO     Building RISC-V instruction generator
Wed, 12 Aug 2020 19:14:40 INFO     Running RISC-V instruction generator
Wed, 12 Aug 2020 19:14:40 INFO     Generating 1 riscv_jump_stress_test
Wed, 12 Aug 2020 19:14:40 INFO     Running riscv_jump_stress_test with 1 batches
Wed, 12 Aug 2020 19:14:40 INFO     Running riscv_jump_stress_test, batch 1/1, test_cnt:1
touch out/seed-1/.metadata/instr_gen.gen.stamp
```

I understand you guys use VCS, I am happy help debugging the issue for Questa.

rswarbrick, 2020-08-12:
Hmm, that's frustrating. Well, there aren't any magic environment variables set in the Makefile, so you should be able to run the `run.py` command above on its own. If you do so, you can add a `-v` flag that turns on "verbose mode". This should give you the Questa command that's actually running, as well as any log output that it generates.

Squinting at the log above, I think you can run:
```
python3 /home/ali/Documents/ibex/vendor/google_riscv-dv/run.py \
     --output=out/seed-1/instr_gen  \
     --steps=gen \
     --gen_timeout=1800 \
     --lsf_cmd="" \
     --simulator="questa" \
     --custom_target=riscv_dv_extension --isa="rv32imc" --mabi=ilp32  \
      --start_seed=1 --test="riscv_jump_stress_test" --testlist=riscv_dv_extension/testlist.yaml --iterations=1 \
     --csr_yaml=riscv_dv_extension/csr_description.yaml --isa="rv32imc" --end_signature_addr=8ffffffc \
     --sim_opts="+uvm_set_inst_override=riscv_asm_program_gen,ibex_asm_program_gen,"uvm_test_top.asm_gen" \
                 +signature_addr=8ffffffc +pmp_num_regions=16 \
                 +pmp_granularity=0 +tvec_alignment=8" \
     -v
```

udinator, 2020-08-13:
Hi @aliess - to add on to what @rswarbrick is suggesting, if the instruction generation still doesn't seem to be working after trying to run the `run.py` command by itself to generate/compile the assembly file, can you please file a duplicate of this issue in https://github.com/google/riscv-dv?
One of our contributors is very familiar with Questa and might be able to provide some more advice (we primarily use VCS to develop the instruction generator).

aliess, 2020-08-18:
I cloned https://github.com/google/riscv-dv and I have the same issue there so I opened an issue there.

rswarbrick, 2020-08-18:
Great, thanks!

imphil, 2020-11-02:
Assuming this is/will be addressed in riscv-dv.

closed 2020-11-02 by imphil

Shaileshkavar, 2022-11-25:
Hello,

When I am trying to run UVM tests, using the below command to generate **riscv_arithmetic_basic_testfor** example:

make instr_gen_run SIMULATOR=questa TEST=riscv_arithmetic_basic_test ISS=spike ISA=rv32imc SEED=1 ISA=rv32imc

**Error**:

build_dir  /users3/shailesh.kavar/riscv-ibex/ibex/dv/uvm/core_ibex/out/build/instr_gen
placeholder_dir  /tmp/tmplp2ba8li/@@PLACEHOLDER@@
build_dir  /users3/shailesh.kavar/riscv-ibex/ibex/dv/uvm/core_ibex/out/build/instr_gen
file_copies [('gen.log', 'gen.log', True), ('test_0.S', 'test.S', False)]
src_path: /tmp/tmplp2ba8li/gen.log
dst_path: /users3/shailesh.kavar/riscv-ibex/ibex/dv/uvm/core_ibex/out/run/tests/riscv_arithmetic_basic_test.10/gen.log

And when I tried to generate the assembly file for the same test in google-riscv, it is generated properly.

[shailesh.kavar@vnc12 ~/riscv-ibex/ibex/vendor/google_riscv-dv]% python3 run.py --test riscv_arithmetic_basic_test --simulator questa   Fri, 25 Nov 2022 11:18:49 INFO     Creating output directory: out_2022-11-25
Fri, 25 Nov 2022 11:18:49 INFO     Processing regression test list : /users3/shailesh.kavar/riscv-ibex/ibex/vendor/google_riscv-dv/target/rv32imc/testlist.yaml, test: riscv_arithmetic_basic_test
Fri, 25 Nov 2022 11:18:49 INFO     Processing regression test list : /users3/shailesh.kavar/riscv-ibex/ibex/vendor/google_riscv-dv/yaml/base_testlist.yaml, test: riscv_arithmetic_basic_test
Fri, 25 Nov 2022 11:18:49 INFO     Found matched tests: riscv_arithmetic_basic_test, iterations:2
Fri, 25 Nov 2022 11:18:49 INFO     Processing simulator setup file : /users3/shailesh.kavar/riscv-ibex/ibex/vendor/google_riscv-dv/yaml/simulator.yaml
Fri, 25 Nov 2022 11:18:49 INFO     Found matching simulator: questa
Fri, 25 Nov 2022 11:18:49 INFO     Building RISC-V instruction generator
Fri, 25 Nov 2022 11:19:04 INFO     Running RISC-V instruction generator
Fri, 25 Nov 2022 11:19:04 INFO     Generating 2 riscv_arithmetic_basic_test
Fri, 25 Nov 2022 11:19:04 INFO     Running riscv_arithmetic_basic_test with 1 batches
Fri, 25 Nov 2022 11:19:04 INFO     Running riscv_arithmetic_basic_test, batch 1/1, test_cnt:2
Fri, 25 Nov 2022 11:19:16 INFO     Compiling out_2022-11-25/asm_test/riscv_arithmetic_basic_test_0.S
Fri, 25 Nov 2022 11:19:16 INFO     Converting to out_2022-11-25/asm_test/riscv_arithmetic_basic_test_0.bin
Fri, 25 Nov 2022 11:19:16 INFO     Compiling out_2022-11-25/asm_test/riscv_arithmetic_basic_test_1.S
Fri, 25 Nov 2022 11:19:16 INFO     Converting to out_2022-11-25/asm_test/riscv_arithmetic_basic_test_1.bin
Fri, 25 Nov 2022 11:19:16 INFO     Processing ISS setup file : /users3/shailesh.kavar/riscv-ibex/ibex/vendor/google_riscv-dv/yaml/iss.yaml