lowRISC/ibex #1078: [formality] Error: Syntax error found in the definition of argument macro `ASSERT`.

islengxw, 2020-08-20:
Hi,
In the formality, we check the RTL and NETLIST function.
Formality (R)  Version O-2018.06-SP5
Any suggestion will be appreciated?
Thanks~

Error: Syntax error found in the definition of argument macro `ASSERT`. (File: /database/projects/rtl/prim_assert.sv Line: 90)  (FMR_VLOG-527)
Error: Syntax error found in the definition of argument macro `ASSERT_NEVER`. (File: /database/projects/rtl/prim_assert.sv Line: 103)  (FMR_VLOG-527)
Error: Syntax error found in the definition of argument macro `ASSERT_KNOWN`. (File: /database/projects/rtl/prim_assert.sv Line: 113)  (FMR_VLOG-527)
Error: Syntax error found in the definition of argument macro `COVER`. (File: /database/projects/rtl/prim_assert.sv Line: 119)  (FMR_VLOG-527)

the error explanation:
fm_shell (setup)> man FMR_VLOG-527
N.  Messages                                                 Command Reference
                                   messages

NAME
       FMR_VLOG-527  (error)  Syntax error found in the definition of argument
       macro `%s`.

DESCRIPTION
       This fatal error occurs when the Verilog analyzer encounters an invalid
       character  in  the definition of an argument macro. OVI1.0 doesn't sup-
       port argument macros. Therefore, you must run cheetah  with  the  -ovi2
       option.

       `define func(a + b) a+b
       module top;
       wire R;
       endmodule

       The macro func definition is wrong in the above example because it uses
       the replaced by the ',' character.

WHAT NEXT
       Correct the netlist.

rswarbrick, 2020-08-20:
It doesn't look like you're using the current head of master (I can tell because `prim_assert.sv` changed quite a bit recently, and these line numbers don't match). Could you try to reproduce with the current version? I don't think this will fix things for you, but it will be much easier to debug at our end.

islengxw, 2020-08-20:
Hi rswarbrick,
I see~ Thank you very much.

islengxw, 2020-08-21:
> It doesn't look like you're using the current head of master (I can tell because `prim_assert.sv` changed quite a bit recently, and these line numbers don't match). Could you try to reproduce with [...]

Hi @rswarbrick , I used the assertion file which is the same as the head of master, the error still there. The point is we didn't encounter this error in the RTL simulation, but in the formaliity step, this error happened when we load the rtl filelist.
In my understanding, we don't need assertion application in the formality step, but in the many rtl files, such as ibex_id_stage.sv will call the `ASSERT, there is no switch to control the assertion implementation.
Do you encounter this error in formality? or consider to improve this?

the error log is ：

Loading include file '/database/projects/rtl/prim_assert.sv'
Loading include file '/database/projects/rtl/prim_assert_standard_macros.svh'
Error: Syntax error found in the definition of argument macro `ASSERT`. (File: /database/projects/rtl/prim_assert_standard_macros.svh Line: 39)  (FMR_VLOG-527)
Error: Syntax error found near ':' in '__name:' <- '  [ unexpected ':', expecting VLOG_DELAY_TIMEUNIT or '#' ]'. (File: /database/projects/rtl/prim_assert_standard_macros.svh Line: 40)  (FMR_VLOG-481)

Thanks
Xiaowei Leng

rswarbrick, 2020-08-21:
It looks like Formality (a tool that I don't believe we've used) doesn't support full SVA syntax. Specifically, it looks like it's choking on named assertions. Fortunately, we can fix this quite easily. If you look at `prim_assert.sv`, you'll see that we include different definitions for macros like `ASSERT, depending on which tool is in use.

Since Formality seems to be a tool for equivalence checking, I don't think you really care about assertions anyway. So please try hacking `prim_assert.sv`, replacing the block of includes at line 70 with just this:
```
`include "prim_assert_dummy_macros.svh"
```

Assuming that works, you'll need to find some way that we can figure out we're running Formality when we're in the preprocessor. For example, does it define a `FORMALITY` macro? Then we can just add lines like
```
`elsif FORMALITY
  `include "prim_assert_dummy_macros.svh"
```
in the middle of the block and all should be well.

zjuwlh, 2020-11-06:
recently  I also have such errors when using formality, using `include "prim_assert_dummy_macros.svh" doesn't fix this issue,
do you still have this issue now? @islengxw

islengxw, 2020-11-13:
> recently I also have such errors when using formality, using `include "prim_assert_dummy_macros.svh" doesn't fix this issue,
> do you still have this issue now? @islengxw

Hi zjuwlh,
We modified the rtl code use '`ifndef SYNTHESIS  ...`endif ' to exclude the assertion.
using `include "prim_assert_dummy_macros.svh" doesn't fix this issue. @zjuwlh

rswarbrick, 2021-01-27:
Since we don't explicitly support the formality tool, and this seems to be a bug in its preprocessor(!), I'm closing this now.

closed 2021-01-27 by rswarbrick