lowRISC/ibex #1083: UVM DV: Return non-zero exit code if a test fails

imphil, 2020-08-20:
Running the regressions like

```
make -C dv/uvm/core_ibex ISS=spike ITERATIONS=1 SEED=1
```

returns a 0 exit code even if a test fails. 

@udinator what do you use to determine test success/failure? Can we modify the tooling to get a non-zero exit code in this case?

udinator, 2020-08-20:
Currently we return 0 all the time - we use the post comparison to determine "true" success/failure.
Sure, I can look into modifying the scripts to return a non-zero code on failures.

imphil, 2020-08-20:
A post comparision script to determine success/failure would also do the trick -- if you have something like that around I can plug it into CI as well.

udinator, 2020-08-20:
The post comparison script I mentioned should already have been plugged into the CI I think, it is the script that provides the final report of how many tests passed or failed.
The return code is actually returned by `sim.py`, so I think any modifications will have to be there.

imphil, 2020-10-23:
@udinator any progress on this? It would be great if CI would not only run tests, but also fail if tests fail :smile:

udinator, 2020-10-23:
Hi @imphil, sorry for the delay on my end!

I have a patch ready to do this, I'll get a PR out and request a review.

closed 2020-10-29 by udinator