lowRISC/ibex #1095: [rtl] PC mismatch alarm caused by dummy instruction insertion

zdogs, 2020-08-24:
Hi,

The PC mismatch alarm fires when a dummy instruction is inserted.

conditions:

- parameter bit SecureIbex = 1'b1,
- Bit 2 of cpuctrl register (dummy_instr_en) is set

[image]

May I suggest following change in ibex_if_stage.sv : 

    // Check that the address equals the previous address +2/+4
    assign pc_mismatch_alert_o = prev_instr_seq_q & (pc_if_o != prev_instr_addr_incr) **& ~dummy_instr_id_o**;

zdogs, 2020-08-24:
Sorry, here are the right waveforms:
[image]

GregAC, 2020-08-27:
Thanks for the report. If you have a fix feel free to submit a PR.

PR #1135 (2020-10-14): [rtl] Various security feature bugfixes
- Document that SecureIbex cannot be used without a multiplier and add
  an assertion in the rtl. This fixes #1080.
- Move the PC checking hardware onto its own parameter to match all the
  other individual security features.
- Make the PC increment behavior more sensible on fetch errors (and make
  it match the icache behavior). Factor this into the PC increment check
  to prevent false triggering, fixes #1094.
- Stop the PC mismatch checker firing on dummy instructions, fixes
  #1095.
commit 8953d82c (2020-10-14): [rtl] Various security feature bugfixes

- Document that SecureIbex cannot be used without a multiplier and add
  an assertion in the rtl. This fixes #1080.
- Move the PC checking hardware onto its own parameter to match all the
  other individual security features.
- Make the PC increment behavior more sensible on fetch errors (and make
  it match the icache behavior). Factor this into the PC increment check
  to prevent false triggering, fixes #1094.
- Stop the PC mismatch checker firing on dummy instructions, fixes
  #1095.