lowRISC/ibex #1101: Failing randomization in ibex_icache_stress_all

dawidzim, 2020-09-07:
Hey!
I have problem with randomization in test `ibex_icache_stress_all` in file `ibex_icache_core_base_seq.sv`:
```
    `DV_CHECK_RANDOMIZE_WITH_FATAL(
       req,

       // Force a branch if necessary
       force_branch -> req.trans_type == ICacheCoreTransTypeBranch;

       // If this is a branch and constrain_branches is true then constrain any branch target.
       (constrain_branches && (req.trans_type == ICacheCoreTransTypeBranch)) ->
         req.branch_addr inside {[base_addr:top_restricted_addr]};

       // If this is a branch and constrain_branches is false, we don't constrain the branch target,
       // but we do weight the bottom and top of the address space a bit higher. This is a weaker
       // version of the weighting that we place on base_addr.
       (!constrain_branches && (req.trans_type == ICacheCoreTransTypeBranch)) ->
         req.branch_addr dist { [0:63]                      :/ 1,
                                [64:32'hffffffbf]           :/ 20,
                                [32'hffffffc0:32'hffffffff] :/ 1 };

       // If this is a branch and constrain_branches is true, we can ask for up to 100 instructions
       // (independent of insns_since_branch)
       (constrain_branches && (req.trans_type == ICacheCoreTransTypeBranch)) ->
         num_insns <= 100;

       // If this isn't a branch and constrain_branches is true, we can ask for up to 100 -
       // insns_since_branch instructions. This will be positive because of the check above.
       (constrain_branches && (req.trans_type != ICacheCoreTransTypeBranch)) ->
         num_insns <= 100 - insns_since_branch;

       const_enable -> enable == cache_enabled;

       // Toggle the cache enable line one time in 50. This should allow us a reasonable amount of
       // time in each mode (note that each transaction here results in multiple instruction
       // fetches)
       enable dist { cache_enabled :/ gap_between_toggle_enable, ~cache_enabled :/ 1 };

       // If must_invalidate is set, we have to invalidate with this item.
       must_invalidate -> invalidate == 1'b1;

       // If no_invalidate is set, we shouldn't ever touch the invalidate line, unless
       // must_invalidate is set (which will only apply once).
       (no_invalidate && !must_invalidate) -> invalidate == 1'b0;

       // Start an invalidation every 1+gap_between_invalidations items.
       invalidate dist { 1'b0 :/ gap_between_invalidations, 1'b1 :/ 1 };

       // If we have seen a new seed since the last invalidation (which must have happened while the
       // cache was disabled) and this item is enabled, force the cache to invalidate.
       stale_seed && enable -> invalidate == 1'b1;

    )
```
problem is with this constraint, I think better way to do this will be change line:
`stale_seed && enable -> invalidate == 1'b1;` to `!(no_invalidate && !must_invalidate) && stale_seed && enable -> invalidate == 1'b1;`
In this way we are sure that this constrain won't be overconstrained

rswarbrick, 2020-09-07:
Are you saying that you have a situation where `no_invalidate` and `stale_seed` are both true at the same time? That's not supposed to happen (and I should probably have added an assertion to check it doesn't). Can you check that's the case?

If it is the case, we'll need to do a bit of digging to see what's gone wrong. I assume you're using Riviera, and I don't have access to a Riviera licence, so I won't be able to reproduce from just the seed. Could you turn on UVM tracing at UVM_HIGH and send a copy of the log file?

dawidzim, 2020-09-07:
Yeah, sure, I sent logs on your e-mail.

rswarbrick, 2020-09-07:
Thanks for that. Could you have a look at the attached PR? I think this should fix things, but it would be nice to see it worked at your end before merging.

dawidzim, 2020-09-07:
```
|             |                | **TOTAL**                         |     9     |    9    |  100.00 %   |
```
Works! Thanks for help :)

PR #1102 (2020-09-07): Fix "no invalidate" logic in ICache sequence
Some sequences run in "no invalidate" mode. Unfortunately, sometimes
we *have* to invalidate, either because we have a stale seed, or
because we're at the start of a sequence.

It turns out that "no invalidate" is a soft requirement, so we can
weaken it here. This patch also renames the flag to
"avoid_invalidation" to make it clear that this is allowed to be
violated occasionally.

Closes #1101.
commit c31d0661 (2020-09-07): Fix "no invalidate" logic in ICache sequence

Some sequences run in "no invalidate" mode. Unfortunately, sometimes
we *have* to invalidate, either because we have a stale seed, or
because we're at the start of a sequence.

It turns out that "no invalidate" is a soft requirement, so we can
weaken it here. This patch also renames the flag to
"avoid_invalidation" to make it clear that this is allowed to be
violated occasionally.

Closes #1101.
closed 2020-09-07 by rswarbrick