lowRISC/ibex #1103: Enable a way to configure the Cache characteristics from the Ibex Core Top

ckdur, 2020-09-08:
Simply I noticed this in the Ibex. The cache is fixed always to:

```
module ibex_icache #(
  // Cache arrangement parameters
  parameter int unsigned BusWidth       = 32,
  parameter int unsigned CacheSizeBytes = 4*1024,
  parameter bit          ICacheECC      = 1'b0,
  parameter int unsigned LineSize       = 64,
  parameter int unsigned NumWays        = 2,
  // Always make speculative bus requests in parallel with lookups
  parameter bit          SpecRequest    = 1'b0,
  // Only cache branch targets
  parameter bit          BranchCache    = 1'b0
)
```

The cache is only instanced in the following line.
https://github.com/lowRISC/ibex/blob/master/rtl/ibex_if_stage.sv#L185
The parameters are not controlled definitely in any external way possible. It would be neat to control the size of the cache from the parameters. Is this the way intended?

rswarbrick, 2020-09-08:
You're right that this isn't currently wired out to the top-level. I think this is mostly because we haven't needed to do so yet. Do you have an application where it would help? If so, can you describe what you'd like to control?

GregAC, 2020-09-08:
I think this is partially to minimise the number of parameters fed out to top-level. Those that are there generally get fed into multiple different places. For the ICache however I think all of those parameters are purely internal to the ICache so you can modify them without effecting any other bits of Ibex (@tomroberts-lowrisc can confirm).

So if you're interested in different cache configurations I'd suggest simply altering the parameters used where the ICache is instantiated, though I'd caution that this may be the only tested parameter set so there may be new bugs to find in different parameter settings.

ckdur, 2020-09-08:
> You're right that this isn't currently wired out to the top-level. I think this is mostly because we haven't needed to do so yet. Do you have an application where it would help? If so, can you describe what you'd like to control?

What I am doing is to instance the Ibex core in the Chipyard multi-core generator. This is for testing OpenTitan in Berkeley's hardware. They have an actual ICacheParams where some of the variables can be directly used from the configurations.

> So if you're interested in different cache configurations I'd suggest simply altering the parameters used where the ICache is instantiated, though I'd caution that this may be the only tested parameter set so there may be new bugs to find in different parameter settings.

I will create a Fork then, and add the parameters directly by simply modifying the code then.

Thank you very much. Will close the issue.

tomroberts-lowrisc, 2020-09-08:
A lot of the parameters in the I$ are fairly specific - but it seems reasonable to expose the cache size at the top level. I'll re-open this issue to track that enhancement.