lowRISC/ibex #1104: compressed Instructions on LSBs

zeeshanrafique23, 2020-09-08:
In```ibex_compressed_decoder``` there is an input ```instr_i``` that is used to detect either the instruction is compressed or not, but WHY ibex always detects the compressed instruction at LSB (```instr_i[1:0])```), although compressed can be at ```instr_i[17:16])```.
```
                    | 31                 16 | 15                  0 |
FIFO entry 0 | Instr 1 [15:0]      | Instr 0 [15:0]     |
FIFO entry 1 | Instr 2 [15:0]      | Instr 1 [31:16]   |
```
How to detect compressed if it is Instr 2.

rswarbrick, 2020-09-08:
The instruction that's being decoded is always at the bottom of the value (so the "am I compressed?" bits are bits 0 and 1). If there is a compressed instruction, the following instruction might appear in the upper half of the value, but it is ignored until the next decode, where it will have been shifted down.

Indeed, the top bits can be completely bogus when the bottom half is a compressed instruction. For example, if we're executing the instruction at `0xfe`, which we loaded by reading 32 bits starting at `0xfc`. We might not actually have the contents of the memory at `0x100` yet. Of course, that doesn't matter because, if the lower half of the value contains a compressed instruction, the decoder can ignore the upper half.

tomroberts-lowrisc, 2020-09-08:
Another way of putting it is - the fetch_fifo abstracts that complexity away from the decoder by always shifting the instruction opcode down to bits [15:0]

zeeshanrafique23, 2020-09-08:
@tomroberts-lowrisc is this the way which is used in ```Ibex_fetch_fifo.sv```? Can you please refer me to the line number of this module.

tomroberts-lowrisc, 2020-09-08:
@zeeshanrafique23 Yep it's at line 84
https://github.com/lowRISC/ibex/blob/c31d066130bdce11dfbacf7afab7641dbd9157f5/rtl/ibex_fetch_fifo.sv#L84
The fetch fifo looks at bit[1] of the address to know when to swap things around.

zeeshanrafique23, 2020-09-08:
Thanks, Tom and  Brick.

closed 2020-09-08 by zeeshanrafique23