lowRISC/ibex #1105: Update fetch_enable signal

tomroberts-lowrisc, 2020-09-08:
We have a requirement from OpenTitan to be able to block the CPU from executing at certain points. I propose we update the existing fetch_enable signal to do that, since it's current behavior is a bit confusing and probably not that useful.

rswarbrick, 2021-01-27:
@tomroberts-lowrisc: Do we still have this requirement? (Since I think we never allow the CPU to recover after it's been squashed, presumably another way to do this on the OT side would be to gate the fetch signal in rv_core_ibex).

tomroberts-lowrisc, 2021-01-27:
Still need to do something here - the fetch_enable_i signal currently does nothing once the core is actually running so would need to change its behaviour and document it

rswarbrick, 2021-01-27:
Understood. What I meant is that we *could* choose just to squash outgoing fetches in `rv_core_ibex`. But, thinking about it, that probably isn't enough when there's also an icache.

commit 6daae350 (2021-06-23): [rtl] Modify fetch_en_i behavior

This signal used to be a one shot enable out of reset. We need an option
to pause execution for OpenTitan, so fetch_enable is extended to cover
that.

The signal is already driven low by the testbench at the end of test.
This is moved after the performance counter reads to ensure they can
complete.

Fixes #1105
closed 2021-06-23 by tomroberts-lowrisc