lowRISC/ibex #1106: The step operation is error when the SW bp is set in a J instruction.

apollojason, 2020-09-10:
Hi Guys： 
      I use the Ibex core running a simple test C code.  
       I set a SW bp before a “jump” instruction,  when the cpu fetched the SW bp instruction, the cpu is halted. 
       But after that, i use the “step" cmd in telnet,  the PC in CPU is wrong, it's not the correct PC which the JUMP instruction expected , the dpc has been changed. 
      When I use the HW bp， the dpc has‘t been changed.
       So, i think this is a bug
       please see the following figure.
      

[image]

[image]

[image]

apollojason, 2020-09-10:
@GregAC Could u help me checking this issue？ Thanks

GregAC, 2020-09-10:
I think this is an issue with Ibex's `EBREAK` handling, it's setting the `depc` for the instruction following the `EBREAK` not the `EBREAK` itself.

From what I can tell from the waveform you've supplied, Ibex is first fetching the instruction immediately following the `jal 0x30000090`, which is the instruction at `0x300006c2` (fetches are aligned so it fetches at `0x300006c0`). The instruction at that address is a 4-byte `EBREAK` (`0x00100073`) which is split across 2 instruction fetches as it's unaligned.

Ibex executes that `EBREAK` giving you `depc = 0x300006c6` which is the address of the instruction after the `EBREAK` where it should produce `depc = 0x300006c2`

From a brief look at the RTL this should work correctly and uses the same mechanism as normal exceptions (and indeed it is tested in the verification environment) so there may be something more complex going on that causes it to break. If you could provide a full .vcd of the failure that would be useful in debugging it.

tomroberts-lowrisc, 2020-09-10:
I would guess that `enter_debug_mode` is set when the core goes to `FLUSH` as a result of the `EBREAK`. That causes the core to go to `DBG_TAKEN_IF` rather than `DBG_TAKEN_ID` hence it captures the wrong PC.

What debug module are you using? and did you definitely clear the HW breakpoint before setting the SW breakpoint?

apollojason, 2020-09-11:
@tomroberts-lowrisc 
 Thanks for your response.

tomroberts-lowrisc:  I would guess that enter_debug_mode is set when the core goes to FLUSH as a result of the EBREAK. That causes the core to go to DBG_TAKEN_IF rather than DBG_TAKEN_ID hence it captures the wrong PC.

apolloJason:  yes, please see the following figure. The SW bp has changed the jal instruction to c.ebreak. So the CPU enter debug_mode and goes to FLUSH, the controller FSM state is DBG_TAKEN_ID, the debug_csr_save is active. But the current PC has been changed in if_stage. The debug_csr_save signal and changed PC caused the wrong dpc in csr register.

tomroberts-lowrisc: What debug module are you using? and did you definitely clear the HW breakpoint before setting the SW breakpoint?

apolloJason: 1. I use the pulp riscv-gdb module, cloned from github.
                     2. I didn't set any HW bp, just use a SW bp. 
                     3. I think the cpu matched a HW bp function is corrected，i have finished a test, just set a single HW bp in the jal instruction, the same PC.
                     4. When I set the SW bp，there is not any HW bp in the CPU trigger.

[image]

apollojason, 2020-09-11:
@tomroberts-lowrisc 
 I set the SW pc in 0x300006c2 ，the J instruction.
 That's a while(1){}; test in the C function.

apollojason, 2020-09-12:
@GregAC @tomroberts-lowrisc 
Hi :
    the cv32e40p has the same bug. They has fixed the bug for storeing the next instruction PC.
    Could you double check again?

GregAC, 2020-09-12:
From the latest wave you've given Ibex is going into `DBG_TAKEN_IF` as `enter_debug_mode` is asserted.

An `EBREAK` doesn't assert `enter_debug_mode` so there's something else going on here as well as the `EBREAK` instruction. Either the debug module has requested to enter debug mode, single stepping is active or there's a hardware breakpoint too.

As I said above a full .vcd would be helpful in debugging (this could be large so gzip it or use .fst which is the GTKWave file format, I think it has a .vcd to .fst conventor available as part of GTKWave).

tomroberts-lowrisc, 2020-09-14:
Single stepping is active, which causes the entry to debug mode. I'm not really sure what the core is supposed to do here (single stepping over an `EBREAK` instruction), it's probably doing the correct thing and stepping to the "next" instruction.

The debugger should replace the `EBREAK` with the real instruction before it releases the core to step. I attempted to reproduce this locally but I don't hit this behavior (I get the correct instruction replaced as expected). You might want to try updating your gdb and OpenOCD versions?

GregAC, 2020-09-14:
Ah had missed the `debug_single_step_o` signal the bottom there.

From the debug spec's description of single stepping:

> If the instruction does not complete due to an exception, the hart will immediately enter Debug Mode before executing the trap handler, with appropriate exception registers set.

`EBREAK` causes an breakpoint exception (if `dcsr.ebreak[m|s|u]` isn't set). So `EBREAK` isn't failing to complete due to an exception, it completes which causes the exception. So I'd argue when single-stepping over an `EBREAK` you should get a `dpc` of the next instruction to execute which is the first instruction of the exception handler.

If `dcsr.ebreak[m|s|u]` are set then I'd argue you'd effectively ignore the single step and enter debug mode as normal for an `EBREAK` when `dcsr.ebreak[m|s|u]` are set (executing the instruction enters debug mode, at which point single stepping would take over but you don't single step in debug mode).

tomroberts-lowrisc, 2020-09-14:
Yeah I think that makes sense.
Looking at the issue raised by @apollojason on cv32e40p, they have changed it such that the `EBREAK` is prioritized here on debug entry rather than the step (thus capturing the ID stage PC rather than the IF stage). We should probably do the same - though only in the `ebreakm/u` case as you noted.

tomroberts-lowrisc, 2020-09-14:
Hi @apollojason can you try again with the linked fix - if it fixes the problem, we can close this issue

apollojason, 2020-09-14:
thank you for explaining it so fully.
 I got sick recently, so i didn't give u a response. Tomorrow, i can get the new version code,try again. 
thanks!

apollojason, 2020-09-23:
@tomroberts-lowrisc @GregAC 
Hi :
     I have try again with the latest version ibex CPU RTL. It is working correctly.
    thank u very much!

closed 2020-09-23 by apollojason