lowRISC/ibex #1113: [dv] ISS support

udinator, 2020-09-22:
Hi all, just wanted to get some feedback before moving ahead to enable the Icache in the Ibex DV environment.

The heart of the matter is that using an Ibex configuration with the Icache enabled means that we have to use a custom branch of Spike (and therefore cannot support OVPsim).
This is because we have to use the custom `cpuctrl` CSR to enable the icache at simulation time - for our purposes I feel it is much more straightforward to add this support in Spike as the simulator code is completely open-source (I've already customized Spike such that it can recognize the Ibex custom CSRs and not throw exceptions on references to them).

Since support for the PMP updates will have to be added to Spike (and since I have another Spike modification in the works to support timer interrupts), I feel like supporting OVPsim may not be possible with these features.

If dropping OVPsim support entirely is not very appealing, a possible solution could be to somehow tie the related Ibex configuration entry to a particular ISS to still support OVPsim with "older" commits of the core.

Any comments/thoughts would be welcome!

rswarbrick, 2020-09-22:
I suggest we do whatever is most convenient for us and ping the OVPSim maintainers to tell them what we've done. If they want to teach their commercial product to work for us, great. If not, no problem.

rswarbrick, 2020-09-22:
That said, doesn't RISC-V allocate a bank of implementation-defined CSRs? If we're just writing to one of those, I'm a bit surprised that the ISS cares: I'd have expected it to just drop the write.

udinator, 2020-09-22:
@rswarbrick thanks for the comments!

> I suggest we do whatever is most convenient for us and ping the OVPSim maintainers to tell them what we've done. If they want to teach their commercial product to work for us, great. If not, no problem.

I agree.

> That said, doesn't RISC-V allocate a bank of implementation-defined CSRs? If we're just writing to one of those, I'm a bit surprised that the ISS cares: I'd have expected it to just drop the write.

Yes, RISC-V does allocate a large space for implementation-defined CSRs; I think accesses to this space are treated as illegal by the ISS because it doesn't actually know about any definitions within this space by default, but the way Spike defines CSRs is very extendable and easy for users (like us) to define implementation-specific CSRs within this space (which is what I took advantage of!).

rswarbrick, 2020-09-23:
Thanks for explaining how we extend CSRs in Spike: that sounds very neat.

imphil, 2020-09-23:
I agree, relying on a modified version of Spike is fine. We just should think about how to make this version available to users. 

1. Create a lowRISC fork repo with the necessary patches?
2. Vendor in spike? (can also be combined with (1))
3. Just have a patch in the tree? (seems fragile)

I'd go for (1). It's not beautiful, but consistent with what we've doing elsewhere and reasonably easy to manage and keep in sync with the RTL, ensuring that users get a version of Spike that works with the state of the Ibex repository they're using.

tomroberts-lowrisc, 2020-09-23:
Agreed - thanks for raising this @udinator 

As for how to make the version available, (2) sounds workable, but I wouldn't be against (1) - we also do that with some other things, and seems more of a route to "upstreaming".

GregAC, 2020-09-23:
Bear in mind we may need a modified spike for ePMP support (which is available here: https://github.com/joxie/riscv-isa-sim).

I think I'd favour 1) as it avoid cluttering up the repo and the current flow for anyone wanting to run the DV environment means they need to acquire spike from somewhere first anyway. So saying it must be from the lowRISC repository isn't a big change to what we have. 2) is a change in the flow, albeit one that makes it easier as you already have the version of spike you need.

GregAC, 2020-09-23:
Thinking about this some more the real question is do we intended to always require a modified version of spike or are we seeing this as a temporary measure until we can get the necessary changes upstreamed (e.g. for the core specific CSR accesses I think there's an argument for a mode in spike that just ignores the writes and always returns 0 for the reads or introducing a config file that allows you to specify a fixed return value for instance)?

If we'll always need a modified version of spike I'd tend towards option 2) otherwise option 1).

rswarbrick, 2020-09-23:
Let's aim for this being a temporary thing (so (1)). Spike already has `dlopen`-based extension support for adding new instructions. If we need it, I'm pretty certain that they'd take a change that let an extension add more CSRs too.

udinator, 2020-09-23:
Thanks all for the ideas!

@imphil thanks for getting an Ibex-specific Spike branch ready. I'll add the existing Spike patch there and work on preparing the second one to merge as well.

imphil, 2020-10-23:
The ibex branch in  https://github.com/lowRISC/riscv-isa-sim now has all necessary patches, and on lowRISC and private CI machines, an appropriate Spike version can be obtained with `module load spike-ibex/20201023-git-255bf1cacc599b1413438c269100f3ecd0eb3352`.

closed 2020-10-23 by imphil