lowRISC/ibex #1114: Where is your interrupt handler code?

zeeshanrafique23, 2020-09-22:
I want to see your exception handler code. Actually I want to know how to write it in assemlby or in C.
I know what to do in the expextion handler (save register, update PC, ...etc). But what to do when you have an exception like misalligned access or illegal csr. So, curious to know.

rswarbrick, 2020-09-23:
The Ibex repository doesn't actually have much software: it just contains the processor itself. However, we do have some software that runs on Ibex in the OpenTitan repository.

@lenary: Could you give zeeshanrafique23 a pointer to where he can find a reasonably full-featured exception handler in OT? (I assume we have one!)

lenary, 2020-09-23:
@zeeshanrafique23 we haven't fleshed out the handlers for that kind of code in OpenTitan yet, but I can give you some tips.

The code for the handlers themselves is in https://github.com/lowrisc/opentitan/blob/master/sw/device/lib/handler.c - and the required declarations are in https://github.com/lowrisc/opentitan/blob/master/sw/device/lib/handler.h - the most important bit is `__attribute__((interrupt))` so that the correct registers are saved and restored by the handler - Clang and GCC will handle this correctly. (If you're only using ibex, `__attribute__((aligned(4)))` is not required, we have it to be safe on other non-rv32imc architectures, and ignore the `__attribute__((weak))`).

These are jumped to by an interrupt vector, which is in https://github.com/lowrisc/opentitan/blob/master/sw/device/exts/common/ibex_interrupt_vectors.S - these use the same names as the declarations in handler.h. The address of this vector (`crt_interrupt_vector`) is the value placed into `mtvec` (this may have been placed into mtvec at reset by setting `boot_address` to that address.

You will see that `handler_exception` does inspect the `mcause` register, I'm not sure we do anything more special than that though - but you can inspect other CSRs to read the illegal instruction address or other pertinent information, as I understand it - this would be done in a way as indicated by the RISC-V specification, ibex doesn't do anything special here.

imphil, 2020-09-23:
There's actually an example for using the timer interrupt in the simple system code, you can find it here at https://github.com/lowRISC/ibex/blob/master/examples/sw/simple_system/common/simple_system_common.c#L192.

zeeshanrafique23, 2020-09-23:
Well, thank you all. Let me review the links you provided. I will definitely continue this thread if got stuck somewhere.
Closing the issue for right now.

closed 2020-09-23 by zeeshanrafique23