lowRISC/ibex #1117: [rtl] Double fault handling

GregAC, 2020-09-24:
Consider the case where an exception occurs on the first instruction of an exception handler (e.g. due to a bus fault or PMP access fault), what should happen?

I suspect Ibex currently enters some kind of infinite loop or deadlocks.

We need to define and document a sensible here behaviour here.

I would suggest some kind of double fault detection that causes the core to halt and assert a top-level signal e.g. `double_fault_o`. System implementers are then left with the choice of what to do (e.g. immediately reset the core). I don't think Ibex should perform a reset itself, simply halt and let the system decide what to do.

rswarbrick, 2020-09-24:
Cores I've worked with in the past just infinite loop. Arguably a reasonable behaviour ("don't set up your software like that!")

lenary, 2020-09-25:
@rswarbrick I want to make sure we don't do anything too silly, but thinking about it, in a double-fault scenario, the watchdog probably would take over, because the core is infinitely looping. That said, a signal probably is a nicer interface than just being entirely unresponsive until the end of time, and having to rely on a watchdog.

rswarbrick, 2021-01-27:
I think we probably need to figure out what we're doing here. As @lenary said, at the moment we presumably just loop on the address until killed by the watchdog. Seems like a perfectly reasonable behaviour to me.

@imphil, @GregAC, @tomroberts-lowrisc: Do we need to think about this further, or can we just consider this closed (possibly with a bit of explicit documentation)?

GregAC, 2021-01-28:
We should see if there's any requirements on the OT side but simply documenting the existing behaviour (i.e. infinite loop) should suffice

GregAC, 2021-04-09:
OT may want some form of double fault detection. We need to define exactly what we consider a double fault (e.g. if you see a synchronous exception in an interrupt handler you may not want to allow the synchronous exception handler to run and wouldn't signal a double fault, but you would for a synchronous exception whilst handling a synchronous exception).

I propose we should build the detector as an independent piece of logic that has no effect on Ibex internally (i.e. no halting on double fault as I initially proposed above). Users of Ibex can observe the double fault and choose to halt the clock, reset the core, raise an NMI etc or ignore entirely, which will result in the logic getting optimised away.

tjaychen, 2021-04-09:
hey all,
From my perspective (@cfrantz @moidx please weigh in), we do not plan any kind of graceful recovery from double fault. 
I believe it is more important we just know "where" we faulted.  So assuming some kind of error caused us to jump into the exception handler, and in there we see any kind of other error, I would think that's a double fault, and "ideally", we would in the crash dump know such an instance occurred while also being made aware of both the faults. 

Specifically to me though, I'm really thinking of the "error and error" cases.  So if we're in an interrupt handler normally, and we see an error there, I would actually consider that a "first" fault.  So once we jump to handle that fault and see another error, then I would consider that a double fault.  I'm not quite sure how this is different from NMIs though... 

@cfrantz is there any thinking from your end you want software to gracefully recover from this? My thinking is no and that we just want blow it up but leave enough breadcrumbs to know where to go and debug.

cfrantz, 2021-04-09:
For my part, I agree with GregAC's proposed definition of double fault: a synchronous exception while handling a synchronous exception.  My preferred response to such a case is to preserve what debug breadcrumbs we can and reset (or alert and allow alert-manager to carry out its defined policy for double-fault).

I'm not sure there can be a graceful software recovery from an error-within-error.  The few times I've had to debug double-faults in the past, the most graceful thing the system could do was drop into the debug monitor so an engineer (me) could look at the stack trace and collect hexdumps.

If we were to go with an alert-manager escalation, we could implement anything "graceful" in the NMI handler, but still allow the escalation to perform a reset should the CPU be unable to respond (ie: bad NMI handler or TL-UL fabric is wedged).

tjaychen, 2021-06-24:
sorry everyone just looping back to this item, since it's been awhile.
It seems we generally agree on the definition as well as the fact there is no need for graceful recovery.

Assuming we have the alert mechanism in place and that causes a reset, would we be able to record enough information in the crash dump on roughly where we last succeeded? Because I imagine if the error we get is simply the exception handler where we faulted again, that may not be THAT useful.  But let me know what you think.

tjaychen, 2021-08-11:
hey all, i just had a conversation with @udinator on this topic and wanted to circle back.

I think whatever @GregAC has proposed sounds good.  Having an explicit, independent detector would also give us the ability to merge this into the alert handling system. 

My remaining question is whether the ibex crash dump output would still be useful in this situation.  Is there any way to feed info about the independent detection back into the crash dump capture so that it holds the initial location of where we error'd and not the subsequent ones?

GregAC, 2021-08-19:
Here's what I'm proposing to implement:

* When any synchronous exception is taken, set a bit
* Clear that bit when `mret` executes
* If a synchronous exception is taken when the bit is set a double fault has been detected
* This bit will be visible and writable in the custom Ibex specific CSR 'cpuctrl'
  - This will allow software to clear it should it want to leave an exception handler without using an `mret`
* Add a new output to ibex `double_fault_seen_o` or similar that will be raised when the double fault is detected

Outstanding questions

1. Do we want a way to disable the detection or the action on detection?
2. How do we want to get the PC of the original exception out to the crash dump? Two similar options
  a. Deal with it internally in Ibex, we'll have two `mepc` fields in the crash dump, one for the original `mepc` one for the second `mepc` when the double fault is seen
  b. Deal with it external to Ibex, when `double_fault_seen_o` is raised we guarantee that cycle the `mepc` in the crash dump is still the original `mepc`. Capture it that cycle so we have both (functionally identical to options 1).

I have a slight preference for option b. for question 2. This is because I'm viewing the double fault handling as a responsibility of the system using Ibex so should keep specific details of extra crash dump information we wish to capture outside of Ibex.

@tjaychen @mundaym @cfrantz any thoughts?

tjaychen, 2021-08-20:
> Here's what I'm proposing to implement:
>
> * When any synchronous exception is taken, set a bit
> * Clear that bit when `mret` executes
> * If a synchronous exception is taken when the bit is set a double [...]

yes this sounds good to me. I'm also fine with dealing with it outside ibex. Since we have the double fault indication, we will probably use it to register the original mepc and also the fact that there was a double fault. The double fault can further be fed into the alert mechanism to cause all sorts of downstream good-ness, and that will give us a clear picture on alert dump as well. 

When you say disable the detection, do you mean disable the double fault detection?
Regarding making the bit clearable by software, is there any way to restrict it to hardware? I'm also worried about software not using mret and forgetting to clear it, thus leading to all sorts of confusion later.

GregAC, 2022-01-24:
Apologies for not answering these initially @tjaychen. Finally getting round to implementing this now.

> Does this mean internally to ibex, you will update the mepc even on double fault, to stay compliant with the spec? Because I think ideally if were to save flops, we would probably not update the mepc based on double fault, but that seems like it would be non-compliant.

Yes, `mepc` updates won't be linked to whether we're seeing a double fault

> When you say disable the detection, do you mean disable the double fault detection?

I was thinking more of the action on detection. Maybe there's a situation where for whatever reason a double fault producing an immediate halt turns out to be a problem? As action of double fault detection is dealt with outside of Ibex we can return to this discussion on the `rv_core_ibex` side in OpenTitan

> Regarding making the bit clearable by software, is there any way to restrict it to hardware?

Certainly, I just felt exposing it to software was better in case we want to do something a bit odd and want to avoid triggering the double fault detection.

For now I'll add two bits to `cpuctrl`. One is the 'synchronous exception seen' bit used to implement the double fault detection and will be read/write. The other will mirror the `double_fault_seen_o` output (read-only). Probably no use on OpenTitan but other users of Ibex may prefer a software solution to double faults. It's easy enough to strip these out or switch to read only later.

tjaychen, 2022-05-28:
@GregAC , have a look at related opentitan issue.
I wonder if we should expose the fact that a synchronous exception was seen from ibex also.
I think originally we used the idea of synchronous exception + mret, as it relates to `mepc` to handle the double fault.

But looking at @jdonjdon 's test, it might be better on synchronous exception to capture the current crash dump.

commit c0f67946 (2022-01-25): [rtl,doc] Add double fault detection.

Fixes #1117
closed 2022-06-16 by tjaychen