lowRISC/ibex #1122: [dv] Functional coverage for Ibex microarchitecture

GregAC, 2020-09-29:
For a while now I've been looking at what we'd like to do for functional coverage for microarchitectural events. I've now come up with a prototype of how we might gather that coverage within the Ibex UVM DV environment.  I've put that prototype on a branch on my Ibex fork and you can see the relevant commit here: https://github.com/GregAC/ibex/commit/cc18a04c66a7ab1d7a8de49c651f63db8724cdad 

I haven't made it a PR as it's a prototype intended to demonstrate how microarchitectural coverage may work rather than a finished product (in particular I've got some notes on things I'd like to cover and there's plenty more to add to this prototype). I've opened this issue to discuss the overall structure.

The general idea is there is a new component, `core_ibex_uarch_fcov` which is instantiated in `core_ibex_env` if an appropriate config parameter from `core_ibex_env_cfg` is set. `core_ibex_uarch_fcov` has a single covergroup which includes various coverpoints and crosses. It samples things from the design via a virtual interface `core_ibex_uarch_fcov_if` which is instantiated in the TB top-level and hierarchical references are used there to probe into the DUT to capture the appropriate signals.
a
Internally to `core_ibex_uarch_fcov` there is a structure `cov_info_t`, which is what the covergroup actually samples and some code in `core_ibex_uarch_fcov` that translate the raw signals from the virtual interface into entries in `cov_info_t`. For example the virtual interface provides the bits of the (decompressed) current instruction in ID, a function in `core_ibex_uarch_fcov` determines an instruction category from these and this is placed in `cov_info_t` and becomes part of the coverage.

Currently the signals in `core_ibex_uarch_fcov_if` are just directly probed signals from the design. For some coverpoints we may need something more sophisticated (e.g. design has been stalled more than 1 cycle, or event X happens the cycle it unstalls). For such things I would suggest the tracking/monitoring that detects these is included either in the design RTL itself (with some suitable macro setup so it's only included for DV runs) or somewhere close to it (module that gets bound in) and then asserts some signal that is then probed by `core_aibex_uarch_fcov_if`. As opposed to doing the tracking/monitoring in `core_ibex_uarch_fcov` itself. The reason for this is such things are very design dependent and will likely break when design changes are made and this breakage may not be immediately obvious. So better they are somewhere that is obvious to whoever is working on the RTL so appropriate changes can be made if RTL behaviour changes. I don't have anything like this in the prototype yet and we can discuss the details of how it's down separately.

The general coverage philosophy will centre around instruction categories. I've only done instruction categories for ID as that's where there's the most variation in design behaviour for different instructions. In IF/WB instructions mostly do the same thing (get fetched/write to the register file) with the few exceptions being interesting events themselves (branch prediction, error response on load/store). So we'll have coverpoints for 'interesting' events and cross that with instruction categories, and potentially cross with ID & WB valid too.

I'd like to approach it from a over-enthusiastic perspective in terms of cover points and cross coverage bins  to start with, see what we actually hit and then make decisions about what out of what is missing is worth covering, adding illegal and ignore bins where appropriate.

Please let me know your thoughts.

udinator, 2020-10-01:
Hi @GregAC, I took a quick look at the linked commit, and I think so far this approach could prove to be very promising.

I agree with the overall structure of the functional coverage prototype so far, and definitely appreciate that it is centering around instruction categories with respect to microarchitectural events in the pipeline. 

RE your comments about the more sophisticated coverpoints (multi-cycle stalls, etc...) it makes sense to me to have the monitoring functionality closer to the RTL, as this seems similar to some features of the implemented performance counters (will there be any overlap?). Like you said though, we can discuss the finer details of this separately.

One question, will this prototype eventually take into consideration interrupts and debug stimuli, correct?

GregAC, 2020-10-01:
> One question, will this prototype eventually take into consideration interrupts and debug stimuli, correct?

Yes definitely, signals indicating events related to those will be added to `uarch_fcov_if` (this may be where we want some multi-cycle tracking, interrupt whilst instruction has been stalled for a few cycles vs interrupt cycle after instruction unstalls for instance). I just didn't want to start adding such things to the prototype in case we decided to go for a different structure.

sriyerg, 2020-10-02:
@GregAC, the solution you've put forth is definitely correct, but...

The alternate approach would be to instantiate all the covergroups inside a module. This could be an independent module that the designer maintains (DV engineer binds it to the DUT in the testbench). The module will only have inputs, which are connected to the internals of the design in the testbench. This eliminates the need for having an interface. It could have helper functions to sample more specific scenarios (as you have done here). Like you said so yourself - it is a lot closer to the RTL, so you can develop the monitoring logic in there to extract more sophisticated coverpoints. 

The module will have bits that can be controlled from the testbench to selectively enable each and every covergroup (in addition to a global disable bit). 

DV engineer can also contribute to this module and choose to not maintain the UVM cov class (unless there is an obvious advantage of doing so; example: being able to look at signals from scoreboard which are at higher levels of abstraction). 

The obvious advantage of this is to allow the designers develop this while being unencumbered from knowing and understanding the UVM testbench (and the ongoing DV progress).  

I will try come up with a prototype before our Tuesday meeting. LMK your thoughts on this.

sriyerg, 2020-10-04:
@GregAC, here's the sample functional-coverage-in-an-interface prototype: 
https://github.com/lowRISC/opentitan/pull/3681

LMKWYT.

rswarbrick, 2020-10-05:
I've just looked through the commit in detail and I like this approach (Greg's). I don't really see much strength in the argument for shielding designers from UVM components. Honestly, they can probably cargo-cult from this example and all will be fine! If, for some bizarre reason, a designer is allergic to UVM components then we could always get someone competent to knock together the boilerplate for them, and then ask them to fill in the signals. (In coloured crayon, perhaps)

My only query is why the `uarch_fcov` interface is connected via hierarchical references, rather than being bound in to the design. Is it just so that you can connect up the writeback stage signal? As we add more "tentacles", I wonder whether it would be cleaner to bind several interfaces in with `.*` and combine their signals one step higher up. It might keep things more structured?

GregAC, 2020-10-05:
Thanks for the prototype @sriyerg 

I do like the way the interface in that prototype gets bound into the design, overall seems neater with less repetition required compared to my prototype though do agree with @rswarbrick that there's no need to worry about shielding designers from the UVM/Testbench side of things to the maximum possible extend. You don't want to require someone to be well practiced in UVM to write some coverpoints but on the other hand there's no harm in having to deal with a few basic things (especially as this can be easily copied from elsewhere).

I do think there's value in the covergroup being instantiated and controlled/sampled within a uvm component because it makes it easier to work with other bits of the testbench and use uvm features if this is useful.

Perhaps a hybrid of the two approaches is called for? Ultimately coverage will be gathered inside a uvm component as with my prototype but we take the binding and interface setup from yours (which I think takes us to something like what @rswarbrick was suggesting by ditching hierarchical references to connect things up and instead using bind).

sriyerg, 2020-10-05:
Actually there are a few more benefits of implementing covergroups in an interface. I have added a [slide](https://docs.google.com/presentation/d/1rUFpWzktORuBZNyG6eSG68kWxEWmxMkMaWdSCadhbQI) now to illustrate them.

vogelpi, 2020-10-06:
Thanks @GregAC and @sriyerg for working out the prototypes. From a designer perspective, I prefer the more streamlined integration proposed by @sriyerg but at the same time I believe that having to learn a little UVM to formulate new coverpoints doesn't hurt. If there are some examples around, this should be doable. Combining the two prototypes using interfaces seems the best solution to me.

One thing I like in particular about @GregAC's proposal is to monitor some advanced conditions directly in the RTL (maybe with a macro to only do it in DV) and then feed the corresponding monitor output to the interface. But if I understand it correctly, this should be possible with both approaches.

simon5656, 2020-10-07:
This is an interesting thread - can somebody point me at the functional specification - as I either cant find it or I dont have access. thanks.

GregAC, 2020-10-07:
Which specification are you referring to? There's the Ibex documentation (https://ibex-core.readthedocs.io/en/latest/) though this does need updating with details of the third pipeline stage (which introduces some interesting behaviour we want to ensure is covered in verification which partially motivates this functional coverage work).

We don't have details of what we'll be covering as part of the functional coverage yet, though I will produce something on that as part of this work.

sriyerg, 2020-10-08:
> Thanks @GregAC and @sriyerg for working out the prototypes. From a designer perspective, I prefer the more streamlined integration proposed by @sriyerg but at the same time I believe that having to [...]

Dang, I really thought "designers not having to learn DV / UVM" was a better selling point for pushing this approach. Please see [slide](https://docs.google.com/presentation/d/1rUFpWzktORuBZNyG6eSG68kWxEWmxMkMaWdSCadhbQI/edit#slide=id.g9e2db54d7b_6_0) explaining several other benefits of implementing functional coverage inside an interface. 

Interface handles are passed to the UVM env like an object. So instead of `ibex_env_cov.foo.sample()`, you would do `ibex_cov_vif.foo.sample()` - you can still write covergroups in the interface that require testbench / scoreboard generated events to be looked at. The UVM testbench still has full control over it. In short, it has all of the same benefits as this approach. 

There are several advantages too - particularly, vertical reuse. At chip level, I can just add the associated bind file as another top level entity and close the "integration" coverage of Ibex, without having to instantiate the entire Ibex UVM env.

GregAC, 2020-10-09:
I'm coming round to @sriyerg's method my main question is what's the most elegant way to deal with signals coming from different modules within Ibex.

For Ibex functional coverage we'll want to probe signals within different modules and bring these together. An individual coverpoint may only ever use signals from a particular module (though I'd prefer this wasn't a hard constraint) and cross coverage will certainly want to cross coverpoints that come from different module signals.

From @sriyerg's prototype we can just use hierarchical references to the various signals in the bind module (which is currently done in #3681) though this does involve some repetition (you need to name the signal in the interface then give a path to it in the bind module). It'd be nice if you could name the signal once in the interface, then use a `.*` in the bind module but then you have the issue that the interface can only pull signals from one module. You need some second level to bring it all together. Maybe we can do something with virtual interfaces and something like looks like `uvm_config_db` but stand-alone and designed specifically for this purpose so there's no UVM dependence or potential to conflict with other parts of the testbench.

I'll have a go at putting something together.

GregAC, 2020-10-09:
I've prototyped something that has a per-module interface where each interface gets bound into its relevant module as with @sriyerg's prototype. However the covergroup doesn't live in the interface. Instead each of the interfaces place a handle to themselves via a static virtual instance variable in a class.

Then in one module I bind in all the interfaces and instantiate the covergroup. It then uses the static virtual instances to refer to the signals it needs to.

It needs a bit of tidying up but once that's done I'll push what I've got to a branch so everyone can take a look. Ultimately it's very similar to @sriyerg's method it's just the covergroup ends up being instantiated in a module rather than an interface. The method may need some tweaking to work in places where we expect multiple instantiations of the same module but that's not an issue for Ibex. Also some annoyances around accessing signals declared in generate blocks still I think it helps cut down on boiler-plate.

GregAC, 2020-10-12:
My second prototype is here: https://github.com/GregAC/ibex/tree/ibex_fcov_v2 which I've described above

The technique it uses may not be great for all OpenTitan IP but could work well for Ibex specifically and indeed wouldn't be incompatible with using @sriyerg's prototype for everything that isn't Ibex.

The main thing it gives you is making wiring up simpler and less error prone as it uses '.*' for wiring in the signals to the interfaces so you only need to write the signal name once. Sadly I can't come up with a way for this to work when you want to get at a signal inside a generate block so you need to wire up things explictly again (see how `wb_valid_q` is wired up in `core_ibex_fcov_wb_stage`).

One downside is this method won't work where you have multiple instances of the same module as they write a handle to themselves to a static class variable in an initial block. However there are various ways this could be dealt with I just don't think it's a concern for Ibex coverage specifically.

sriyerg, 2020-10-12:
Looks fine to me, if you expect there will be lots of signals picked up from various sub-modules combined into the same covergroup. You will need to provide a yet another interface for controls to enable / disable the covergroups.

rswarbrick, 2021-01-25:
@GregAC: Does this issue need to remain open, or is it moot now that the initial functional coverage PR has been merged?

GregAC, 2021-01-27:
We can close this

GregAC, 2021-01-27:
Functional Coverage PR can be found here: https://github.com/lowRISC/ibex/pull/1239

There are still coverpoints to add but it gives the framework used to gather coverage.

closed 2021-01-27 by GregAC