lowRISC/ibex #1129: Illegal Instruction Exception not Raised

shetalani, 2020-10-13:
**RISC-V Specification:**

> Unprivileged ISA

**Issue Description:**

An illegal instruction updates the register file and the associated csrs and executes normally without raising an illegal instruction exception.

**Example:**

The instruction 32'h2820d013 is decoded and executed with no illegal instruction exception being flagged, as machine csrs: mepc, mstatus and mcause aren't updated. 

[image]

---

Product: OneSpin 360 DV-Verify
App: Processor Verification App
Tool's version: 2020.2.0

tomroberts-lowrisc, 2020-10-13:
Thanks for raising this @shetalani 
I have a proposed fix in #1130 - can you give that a go and let us know if it fixes the problem?

shetalani, 2020-10-13:
@tomroberts-lowrisc It is fixed, my pleasure!

closed 2020-10-14 by tomroberts-lowrisc