lowRISC/ibex #1133: Software Interrupts For Machine Mode

sajjadahmed677, 2020-10-14:
I am looking for interrupt handling and CSR implementation in my design as i have only machine mode(for reference i am looking into ibex core) so, my question is what type of software interrupts we handle when CORE only has machine mode.

tomroberts-lowrisc, 2020-10-14:
Hi @sajjadahmed677 - I think just the single machine-mode software interrupt, controlled by the `mip.MSIP` and `mie.MSIE` bits as Ibex has.

You might want to look at the CLINT specification for an example of how the software interrupts might be generated - there is an example implementation here: https://github.com/pulp-platform/clint

sajjadahmed677, 2020-10-15:
thank you @tomroberts-lowrisc i will look into it, and then come back to you if i have any query. actually my design is bare-metal i am using cross compilation for simple programs that's why i am confused with software interrupts. one more thing i have PLIC in my design so if you have material for PLIC that how it generates or handles software interrupts please suggest me.

tomroberts-lowrisc, 2020-10-20:
Hi @sajjadahmed677 
I don't believe the PLIC has any support for software interrupts.

sajjadahmed677, 2020-10-26:
Thank you @tomroberts-lowrisc  for your support.

closed 2020-10-26 by sajjadahmed677