lowRISC/ibex #1138: purpose of ibex_csr file and user mode csrs

sajjadahmed677, 2020-10-15:
i am looking into the CSR implementation done in ibex core. just seen the ibex-csr file which is for csr primitives. will anyone please elaborate this file and its purpose when there is complete csr file present.
also i haven't seen the user mode csrs though ibex supports user mode also as per my knowledge. so, how it handles the user mode when the user mode csrs are not present in csr list.

tomroberts-lowrisc, 2020-10-16:
Hi @sajjadahmed677 the `ibex_csr.sv` file is just a convenient way of adding security hardening features to individual csrs in a consistent way. It is instantiated in the `ibex_cs_registers.sv` file.

Which user mode csrs are you referring to?

sajjadahmed677, 2020-10-16:
thank you @tomroberts-lowrisc 
i am asking about the user mode csrs like mstatus has a secure variant ustatus for user mode and other csrs uie, utvec etc so where are these csrs. or how ibex manage user mode without them.

tomroberts-lowrisc, 2020-10-16:
Hi @sajjadahmed677 - those CSRs are only required with the N extension (U-mode traps/interrupts) which Ibex does not support

sajjadahmed677, 2020-10-17:
Thank you @tomroberts-lowrisc

closed 2020-10-17 by sajjadahmed677