lowRISC/ibex #1139: riscv_debug_branch_jump_test case fail

zjuwlh, 2020-10-16:
Hi,

when I run the riscv_debug_branch_jump_test case , the case will fail because access the unintialzed address, can anyone also find the same issue? Or can anyone help check this issue? my command is the following:
**make ISA=rv32imc ITERATIONS=1 IBEX_CONFIG=small TEST=riscv_debug_branch_jump_test SEED=29354**
can someone also help run on your side to check if there is the same issue on your side?

Thanks,
Lihua

zjuwlh, 2020-10-16:
by the way, I use the OVPSIM as a ISA.

tomroberts-lowrisc, 2020-10-16:
Confirmed I also see the same issue. I think there is a known problem with this test at the moment - @udinator might know?

udinator, 2020-10-16:
Yes, there is a known issue with the test at the moment.

zjuwlh, 2020-10-19:
Thanks for your confirm, do you have any plan when will improve the test?

tomroberts-lowrisc, 2020-10-20:
I have a fix for the issue [here](https://github.com/google/riscv-dv/pull/734)

udinator, 2020-10-26:
This issue should now be resolved - the updates to the instruction generator have been vendored into Ibex repository.

closed 2020-10-26 by udinator