lowRISC/ibex #1143: Run more Ibex configurations in private CI

imphil, 2020-10-23:
Currently we run the following command in private CI: `make -C dv/uvm/core_ibex ITERATIONS=1 TEST=riscv_rand_instr_test,riscv_multiple_interrupt_test,riscv_debug_basic_test ISS=spike`.

This, AFAIK, runs the default configuration. We should at least run a version which is close to what we're using in OpenTitan, and what we aim to be using in OpenTitan. 

Which configurations would that be, and what does the invocation look like? Pass `IBEX_CONFIG` to make?

@GregAC @tomroberts-lowrisc @rswarbrick @udinator

GregAC, 2020-10-23:
The default config is actually 'experimental-maxperf-pmp-bmfull', i.e. the OpenTitan one. Passing an 'IBEX_CONFIG' parameter to make will allow you to choose a different configuration.

Though I would caution this mostly works but can sometimes break. The issue lies in the bitmanip instructions as the test framework doesn't understand that when the bitmanip config is disabled it shouldn't generate bitmanip instructions. However bitmanip instruction generation is confined to certain tests (I think), so you can avoid the issue by only running specific tests which is already being done with your command. This issue is a good reminder to me I need to look into this and file an issue/suggest a change for RISCV-DV if required.

I'd suggest we run all of the named configs in the README other than "micro", i.e. "small", "experimental-maxperf" and "experimental-maxperf-pmp-embfull". From what I remember RISC-DV has issues with the reduced register set on "micro" but we should look into that.

GregAC, 2020-10-23:
Oh and maybe add one of the PMP tests to the "experimental-maxperf-pmp-bmfull" run

udinator, 2020-10-23:
I agree with Greg, I think it's a good idea to add one PMP test and one bitmanip test to the private CI run of the `experimental-maxperf-pmp-bmfull` config.

> Though I would caution this mostly works but can sometimes break. The issue lies in the bitmanip instructions as the test framework doesn't understand that when the bitmanip config is disabled it [...]

Yeah, this is an issue with how we generate illegal instructions in riscv-dv. I've started working on a patch for this, but it's low-ish priority on my end as of right now - I can prioritize this if you feel it's necessary sooner rather than later?

> I'd suggest we run all of the named configs in the README other than "micro", i.e. "small", "experimental-maxperf" and "experimental-maxperf-pmp-bmfull". From what I remember RISC-DV has issues with the reduced register set on "micro" but we should look into that.

RISCV-DV actually doesn't support the E extension at all, but like before this is something we can move up the TODO list if you feel it's necessary soon?

imphil, 2020-10-27:
I've now enabled the same configurations we have in public CI also in private CI:

- small
- experimental-maxperf-pmp
- experimental-maxperf-pmp-bmfull
- experimental-maxperf-pmp-bmfull-icache
- experimental-branch-predictor

For all these tests, we run the sanity regression set, e.g. `make -C dv/uvm/core_ibex ITERATIONS=1 TEST=riscv_rand_instr_test,riscv_multiple_interrupt_test,riscv_debug_basic_test ISS=spike IBEX_CONFIG=experimental-maxperf-pmp-bmfull-icache`. All tests currently pass.

imphil, 2020-10-27:
@udinator @GregAC So the plan would be to always run the set of riscv_rand_instr_test,riscv_multiple_interrupt_test,riscv_debug_basic_test, and additionally run some tests for the PMP and icache configurations? Which tests should that be? 

If we can make it work with a single version of Spike, that's preferred. If not, we can run the tests with different builds of Spike, that shouldn't be a blocker. Just let me know how you'd like Spike configured.

udinator, 2020-10-27:
@imphil yup, I think that is the plan. For PMP, we can run the `riscv_pmp_full_random_test`, and to specifically try and stress the icache, we can run the `riscv_rand_jump_test`. 
A single version of spike will work for all of these tests, so there's no need to try and configure different versions for different tests.

imphil, 2020-10-27:
@udinator great, I'll add these tests to CI tomorrow and let you know how it goes.

udinator, 2020-10-27:
sgtm!

GregAC, 2021-12-10:
We now run multiple configurations in private CI

closed 2021-12-10 by GregAC