lowRISC/ibex #1148: mstatus register width

sajjadahmed677, 2020-10-26:
guide me about the reason behind setting the width of mstatus csr to the width of "status_t struct" rather than having 32bit width. as described in privilege spec.

imphil, 2020-10-26:
The mstatus register is 32 bit wide from a software point of view, as described at https://ibex-core.readthedocs.io/en/latest/03_reference/cs_registers.html?highlight=mstatus#machine-status-mstatus and matches what is specified in the RISC-V Privileged Specification. Everything else is just an implementation-detail, where we skip bits which are not relevant to our implementation.

vogelpi, 2020-10-26:
Hi @sajjadahmed677 ,
the difference in width of the implemented register and the description in the privileged spec (Section 3.1.6) comes from the fact that not all bits of this register have to be implemented to be spec compliant.

For example, there are several WPRI fields in `mstatus` where WPRI means reserved writes preserve values, reads ignore values. As explained in Section 2.3 of the privileged spec, such bits/fields are reserved for future use and further: "For forward compatibility, implementations that do not furnish these fields must hardwire them to zero." Consequently, it does not make sense for Ibex to implement those fields with registers that always read as 0 and cannot be written. It would be a waste of resources.

Similarly, there are bits that only matter if also U- and S-modes are implemented like (`SPIE`, `UPIE`) etc. These bits should always read as 0 and are not writable if a core doesn't implement U- and S-modes like Ibex.

Giving `mstatus` the width of the `status_t` struct achieves exactly that: The `status_t` struct defines which bits/fields are implemented using real registers. The rest is always 0.

sajjadahmed677, 2020-10-26:
Alright got i the point. Thank you @vogelpi and @imphil for such detailed description.

closed 2020-10-26 by sajjadahmed677