lowRISC/ibex #1154: [dv] Regression failure for riscv_multiple_interrupt_test with config experimental-branch-predictor

imphil, 2020-10-27:
* Seed: 3116
* `make -C dv/uvm/core_ibex ITERATIONS=1 TEST=riscv_rand_instr_test,riscv_multiple_interrupt_test,riscv_debug_basic_test ISS=spike IBEX_CONFIG=experimental-branch-predictor`
* CI run: https://dev.azure.com/lowrisc/lowrisc-private/_build/results?buildId=24876&view=logs&j=89b1484e-e59b-56b8-bef8-ec534b74be7c&t=69a04726-f3d2-5f59-d3f4-a33300a2ed9f
* Full build log in CI: https://artproduks1.artifacts.visualstudio.com/Ace3fc167-bb91-4c33-a65a-ae648df7f974/4efdbdf8-a184-4518-9864-57d2a49c2aed/_apis/artifact/cGlwZWxpbmVhcnRpZmFjdDovL2xvd3Jpc2MvcHJvamVjdElkLzRlZmRiZGY4LWExODQtNDUxOC05ODY0LTU3ZDJhNDljMmFlZC9idWlsZElkLzI0ODc2L2FydGlmYWN0TmFtZS9vdXQ1/content?format=file&subPath=%2Fout.tar.xz

```
Tue, 27 Oct 2020 15:01:21 ERROR    Timeout[300s]: env SIM_DIR=out/seed-3116/rtl_sim/riscv_multiple_interrupt_test.0   out/seed-3116/rtl_sim/vcs_simv +vcs+lic+wait +signature_addr=8ffffffc  -ucli -do ./vcs.tcl       +ntb_random_seed=3116 +UVM_TESTNAME=core_ibex_debug_intr_basic_test +bin=out/seed-3116/instr_gen/asm_tests/riscv_multiple_interrupt_test_0.bin     +ibex_tracer_file_base=out/seed-3116/rtl_sim/riscv_multiple_interrupt_test.0/trace_core     -l out/seed-3116/rtl_sim/riscv_multiple_interrupt_test.0/sim.log  +require_signature_addr=1 +enable_irq_multiple_seq=1 
```

@udinator @GregAC is this something you can have a look at? The logs should contain all necessary information.

udinator, 2020-10-27:
Sure, I will take a look into this failure today.

udinator, 2020-10-27:
@imphil, I'm unable to recreate a failure here locally with SEED=3116, and I don't have access to view the CI log files it seems.

The `riscv_multiple_interrupt_test` with the `experimental-branch-predictor` config did take close to 3.5 minutes on my local machine though, perhaps increasing the timeout period for RTL simulation might help? This can be done by going to the `run_sim_commands` function inside of `sim.py`, and modifying the value `300` that is passed into the `run_cmd` invocation.
(Now that I think about it it might be good to have this timeout be controllable from cmdline...)

udinator, 2020-10-28:
As far as I can tell from the output logs nothing out of the ordinary happened during simulation, the test just seemed to have timed out since it was running for over 300 seconds.
As a result I think #1155 should prevent this sort of error from popping up in the future.

imphil, 2020-10-29:
Thanks a lot for checking @udinator. Let's close this issue then and hope it doesn't reappear! :smile:

closed 2020-10-29 by imphil