lowRISC/ibex #1168: riscv_irq_in_debug_mode_test fail

zjuwlh, 2020-11-02:
Hi

When IBEX_CONFIG=small, SEED=27754, riscv_irq_in_debug_mode_test fail because of test bench doesn't capture the "dret" instruction, could you please help check if it is really a RTL issue? since from waveform, I see some 'dret' flag asserted

Thanks,
Lihua

GregAC, 2020-11-02:
Hi @zjuwlh thank you for the bug reports.

It would be helpful if you could include a log file from the test in each case along with the tools you're using. Whilst the seed can help reproduce we may see different test behaviour between different tools and tool versions so it's not a guaranteed way to reproduce. By seeing the log files we can better understand the issue and be more confident we're reproducing what you're seeing.

GregAC, 2020-11-02:
I was unable to reproduce this (everything passes for me). There was an odd issue with ovpsim timing out in some tests but it appeared to be caught in some kind of infinite loop at the end of the program which didn't actually cause an issue for the post sim comparison (the instructions before the loop at the end being the important part).

GregAC, 2020-11-02:
@zjuwlh for those issues I couldn't reproduce (this one and https://github.com/lowRISC/ibex/issues/1166) could you ensure you're running with the latest version of the Ibex repository along with supplying the other info I mentioned above?

zjuwlh, 2020-11-03:
OK, I will try the latest version.

zjuwlh, 2020-11-03:
this test is pass in the latest database version, please close this issue, thanks.

closed 2020-11-03 by GregAC