lowRISC/ibex #1169: Running single test case using Xcelium

rahulraveendran15-coder, 2020-11-02:
Hi,
I just want to run a single test case using Xcelium simulator, I can see that the support has been added for xcelium.
Could you please guide me or help me in running a testcase with xcelium and the proper command to run it.

I am new to the platform even I dont know the command single test case with VCS.

rswarbrick, 2020-11-02:
Hi there!

You'll find a description of the UVM-based testbench for Ibex in the [reference section](https://ibex-core.readthedocs.io/en/latest/03_reference/verification.html) of our documentation (which is not a great place for this - I'll file an issue to move it!). I'd suggest you have a read through that for an overview of how everything works. To run some tests with Xcelium, the basic command is something like
```
cd dv/uvm/core_ibex
make SIMULATOR=xlm
```
but there are a couple of things to bear in mind.

Firstly, you need a special version of Spike (the ISS) for some of the tests to pass. This is because the upstream version doesn't really support how our icache works. We'll upstream our patches to get this sorted eventually, but for now you need a "slightly forked" version. You can find the code at https://github.com/lowRISC/riscv-isa-sim/tree/ibex (this is technically explained in the documentation that I linked to, but it's reasonably easy to skip over by accident).

Secondly, you'll need a RISC-V toolchain that supports the bitmanip extension. See the documentation for details (I suggest the pre-built toolchain).

Thirdly, the whole flow is not massively robust. If you see something going awry or something just looks obviously silly, please file an issue! It might be a quick fix, and I'd love to get this tidied up a bit. If you're stuck, you could also reach out to us on Zulip (for details, see the ["community" page](https://www.lowrisc.org/community/) on the lowRISC website): we might be able to help you quicker there.

Hope this helps,

Rupert

rahulraveendran15-coder, 2020-11-02:
> Hi there!
>
> You'll find a description of the UVM-based testbench for Ibex in the [reference section](https://ibex-core.readthedocs.io/en/latest/03_reference/verification.html) of our documentation [...]

Hi, 
Thank you for the support,
I have two more concerns:

# Run a single test
1. make TEST=riscv_machine_mode_rand_test ITERATIONS=1

# Compile and run RTL simulation
2. make TEST=xxx compile,rtl_sim

Question 1: What is the difference between both these commands if I am running a single test will it be a RTL simulation?
Basically I am not getting the difference between both.

Question 2. What is the use of sim.py in path dv/uvm/core_ibex we have already a Makefile right?

Question 3:  but there are a couple of things to bear in mind. from your three points are these things only need to do for Xcelium simulator or for all simulators?

rswarbrick, 2020-11-02:
For question 1, the example in the documentation that gives your second command is supposed to show how to use the phony targets in the Makefile to decide what to do (build the testbench, generate instructions, simulate stuff, etc.). If you don't pass a target at all, the Makefile defaults to building and running the tests (which is probably what you want). That example is actually bogus (we have `compile` and `rtl_sim` targets, but not one called `compile,rtl_sim`). I'll file a follow-up to sort that out, thanks.

For question 2, `sim.py` is essentially just a nice frontend that avoids us having to put all the logic about "how do you run this EDA tool?" in the Makefile. Python is a **far** better language for that sort of thing! It also handles a bit of CSV parsing, if I remember correctly.

For question 3, my comments about the ISS and toolchain are independent of your choice of RTL simulator.

For question 4: All of the tests work by generating some assembly, building it and then running on an instruction set simulator (ISS) and the RTL, checking that the two behave the same[*]. So you need both for all the tests, I think.

[*] That's not strictly true for some of the debug tests, but you do at least need an ISS.

rahulraveendran15-coder, 2020-11-02:
> For question 1, the example in the documentation that gives your second command is supposed to show how to use the phony targets in the Makefile to decide what to do (build the testbench, generate [...]

Hi,

Thank you very much for the detailed explanation.
in the last point you have mentioned that "building it and then running on an instruction set simulator (ISS) and the RTL, checking that the two behave the same[*]" What is this "two behaves the same" corresponds to?

Also in my view the I though that the "C" test cases are compiled to assembly and then loaded the corresponding binary to memory and perform the simulation. This is with use of SDK .
But the test cases in DV folder here are UVM test cases so are we converting these testcases also into assembly language of (RISCV ) format and putting the corresponding binary file in to memory

rswarbrick, 2020-11-02:
I'm not sure you have read [the documentation](https://ibex-core.readthedocs.io/en/latest/03_reference/verification.html) that I linked to? In particular, make sure you understand the text around the diagram called "End-to-end RTL/ISS co-simulation flow". That should hopefully clear things up for you.

rahulraveendran15-coder, 2020-11-02:
> I'm not sure you have read [the documentation](https://ibex-core.readthedocs.io/en/latest/03_reference/verification.html) that I linked to? In particular, make sure you understand the text around [...]

Yes, now I am geeting a  more clear picture but I think apart from the two points you mentioned in the first solution I think we need "https://github.com/google/riscv-dv" clone also to generate random instructions am I right.
So

1.  riscv-dv
2. ISS (Spike)
3. GCC-TOOL chain
These 3 are needed along with ibex

udinator, 2020-11-05:
Hi @rahulraveendran15-coder - the riscv-dv instruction generator is a part of the overall Ibex simulation flow, but you do not need to install this yourself as it is already included in this repository when running simulations.

As @rswarbrick has explained, you will only need to manually install the Spike ISS found at https://github.com/lowRISC/riscv-isa-sim/tree/ibex as well as a RISC-V toolchain that supports bitmanipulation extension (the prebuilt toolchain is recommended).

rahulraveendran15-coder, 2020-11-06:
Hi @udinator 
Thank you very much for explaining.
1. Can I use OVPsim simulator instead of spike because spike is causing me some issues?

2. Why do the  tool chain requires bit manip extension? I  am planning to use rv32imc instruction set architecture in my project. So why do I need really need bit manip prebuilt in  tool chain?
3. Is it because some of the test cases in ibex will fail if i dont use the bit manip supported ? 
4. Can I use the normal riscv linux tool chain Build it and use instead of bit manipulation?

GregAC, 2020-11-06:
> Can I use OVPsim simulator instead of spike because spike is causing me some issues?

We needed a custom fork of spike to support writes to a custom CSR that enables and disables the ICache. On my machine it  I can run the latest Ibex master DV against OVPSim so it looks like OVPSim is happy with this custom CSR usage and can be used @udinator may know more.

> Why do the tool chain requires bit manip extension? I am planning to use rv32imc instruction set architecture in my project. So why do I need really need bit manip prebuilt in tool chain?

Bitmanip is not required, you will however need to remove any bitmanip tests (see https://github.com/lowRISC/ibex/blob/master/dv/uvm/core_ibex/riscv_dv_extension/testlist.yaml you would remove the bitmanip tests from there).

> Is it because some of the test cases in ibex will fail if i dont use the bit manip supported ?

Yes, see above, these tests can be removed

> Can I use the normal riscv linux tool chain Build it and use instead of bit manipulation?

A 'riscv-unknown-linux-' or other 'riscv' tool chains that are built for linux may not work so well for Ibex. The toolchain assumes the existence of a linux system but there it is running on bare-metal. I would advise using a 'riscv-unknown-elf' toolchain (whether one from our toolchain repository or elsewhere).

udinator, 2020-11-10:
@GregAC the issue you brought up is addressed by #1184.

rswarbrick, 2021-01-25:
@rahulraveendran15-coder: I'm closing this issue now because, as far as I can tell, all your questions have been answered. Please re-open if you think there's something we've missed.

closed 2021-01-25 by rswarbrick