lowRISC/ibex #1173: ISS Simulation and GNU Tool chain setup ambiguities

rahulraveendran15-coder, 2020-11-03:
HI,
There are some ambiguities in the user manual of ibex verification environment setup
(1) A RISC-V instruction set simulator, such as Spike or OVPsim the links are not broken here
(2) I have gone through the Spike installation link "https://github.com/lowRISC/riscv-isa-sim/tree/ibex", but in this  its not mentioned anything like " Note that Spike must be configured with --enable-commitlog and --enable-misaligned." in the manual. What does this corresponds to?
(3) Spike does not support the RISC-V Bit Manipulation Extension (Bitmanip) by default. To support this draft extension implemented in Ibex, the **riscv-bitmanip branch** of Spike needs to be used. The link is broken and the corresponding page is not found. How to use this ?
(4) For the tool chain I am building it because the I cannot use the prebuilt tool chain since I am using centos 7 and the glibc is 2.17 which cannot be updated to 2.7. After setting the tool chain how to manually install the bit manipulation patches in it I didnt get much information from the links provided.

GregAC, 2020-11-03:
> A RISC-V instruction set simulator, such as Spike or OVPsim the links are not broken here

The OVPsim link works for me, however the Spike one is broken as it points to the bitmanip branch, thanks for letting us know I'll get it fixed.

>  I have gone through the Spike installation link "https://github.com/lowRISC/riscv-isa-sim/tree/ibex", but in this its not mentioned anything like " Note that Spike must be configured with --enable-commitlog and --enable-misaligned." in the manual. What does this corresponds to?

These are options to pass to the configure script e.g.

```
./configure --enable-commitlog --enable-misaligned
```

This isn't hugely clear from the documentation so I'll add an clarification.

> Spike does not support the RISC-V Bit Manipulation Extension (Bitmanip) by default. To support this draft extension implemented in Ibex, the riscv-bitmanip branch of Spike needs to be used. The link is broken and the corresponding page is not found. How to use this ?

It looks like the branch has been retired and bitmanip support is merged into the master branch so normal spike should be sufficient (though I haven't yet tested this myself). Though I'd note the bitmanip spec hasn't been ratified yet and they're currently looking at various tweaks for the final v1.0 spec. So it's likely there will be some changes to the spike master branch that don't match how we've implemented bitmanip. I'd suggest using our fork of spike for now to avoid this issue.

> For the tool chain I am building it because the I cannot use the prebuilt tool chain since I am using centos 7 and the glibc is 2.17 which cannot be updated to 2.7. After setting the tool chain how [...]

Afraid I don't know anything about toolchain builds, I think this is best discussed in your issue on the toolchain repository here: https://github.com/lowRISC/lowrisc-toolchains/issues/36

rahulraveendran15-coder, 2020-11-03:
Hi,
THank you for the answers

I need a bit more clarification

(1) I'd suggest using our fork of spike for now to avoid this issue. >> I didnt get this, because you told that the functionality is merged with master branch. SO can I checkout the master branch?
I have done some thing like this 
git clone https://github.com/lowRISC/riscv-isa-sim/tree/master
then done a 
git checkout ibex

Is this the one you are asking to do by this point  "I'd suggest using our fork of spike for now to avoid this issue" or else could you please explain what you meant by this?

(2) Can I do something like "../configure --prefix=$RISCV --enable-commitlog --enable-misaligned " as it is mentioned in the build step os SPike ISS?

GregAC, 2020-11-03:
> I have done some thing like this
git clone https://github.com/lowRISC/riscv-isa-sim/tree/master
then done a
git checkout ibex

> Is this the one you are asking to do by this point "I'd suggest using our fork of spike for now to avoid this issue" or else could you please explain what you meant by this?

Yes that gets you our fork of spike

> Can I do something like "../configure --prefix=$RISCV --enable-commitlog --enable-misaligned " as it is mentioned in the build step os SPike ISS?

Yes that's what I meant

rahulraveendran15-coder, 2020-11-03:
THank you very much for the confirmation.

So I think in the ibex branch of Spike both the functionalities
 (1)Bit manipulation extension and
(2) Instruction cache has been enabled. 
Both these functionalities are in the ibex branch am I right?

If I am right the point in the user manual "To support this draft extension implemented in Ibex, the riscv-bitmanip branch of Spike needs to be used. If it is desired to simulate the core with the Icache enabled, a lowRISC-specific branch of Spike must be used, found here." makes confusion to user like thinking to use different branches(even though the first retired branch link is broken) for these two different functionalities

GregAC, 2020-11-05:
> Both these functionalities are in the ibex branch am I right?

Correct, I will update the documentation to make this more clear. As you say the current wording is a little confusing.

GregAC, 2020-11-13:
Document fixes mentioned have been done in https://github.com/lowRISC/ibex/pull/1190 and otherwise I think we've resolved everything else in this issue.

closed 2020-11-13 by GregAC