lowRISC/ibex #1174: [DV] SV disable fork label - Xcelium simulator differences

PaulOKCrevinn, 2020-11-03:
While running and debugging Ibex regression we came across some differences in how Cadence Xcelium simulates disable of fork labels. 
We've locally fixed this issue in 6 different testbench files and we can submit the change requests ourselves,
I wanted to raise an issue on this as it is something that needs to be monitored in the future. 
I am not sure if the testbench code is linted, but is it possible to lint for the disable label construct and flag it in future? 

Below is a summary of the issue we've found, we will follow up with the changes to the Ibex git repo this week. 

**Issue: Disabling a fork process by label name.** 
Cadence support site recommends that when a fork is to be disabled it is to be disabled using the "disable fork" statement and not the "disable fork_process_label" construct, 
the reason being that the "disable fork_process_label" construct for forks is not defined in the IEEE Std 1800-2017 LRM. See section 9.6.2 of the LRM (2017). 
As far as we have seen Cadence Xcelium will ignore a "disable fork_process_label" line and continue, thereby leaving the process running and altering the function of the testcase. 
This cause multiple test failures in a Ibex Xcelium regression and lead to particular problems with the riscv_reset_test. 

So for example in the following Ibex code in core_ibex_test_lib.sv: The "**disable imem_fork**" should be replaced with "**disable fork**" in order to be compatible with Cadence Xcelium. 
We have done very limited testing on these changes on EDA playground for Cadence Xcelium 19.09, Mentor Questa 2020.1, and Synopsys VCS 2019.06 and the "disable fork" code seem executes correctly for all these simulators. 

          fork : imem_fork
            begin
              check_next_core_status(HANDLING_EXCEPTION, "Core did not jump to exception handler");
              check_priv_mode(PRIV_LVL_M);
              latched_imem_err = 1'b1;
              `uvm_info(`gfn, $sformatf("latched_imem_err: 0x%0x", latched_imem_err), UVM_LOW)
            end
            begin
              clk_vif.wait_clks(750);
            end
          join_any
          **disable fork;    //  (Changed from:   disable imem_fork;)** 

Cadence support ticket referencing the above issue (NOTE: you will need a Cadence support account to view these): 

https://support.cadence.com/apex/ArticleAttachmentPortal?id=a1Od0000000syTQEAY&pageName=ArticleContent

Other Cadence support issues that deal with fork-join disable behaviour: 

https://support.cadence.com/apex/ArticleAttachmentPortal?id=a1O0V000006DeTOUA0&pageName=ArticleContent 

https://support.cadence.com/apex/ArticleAttachmentPortal?id=a1Od0000000uSMiEAM&pageName=ArticleContent

imphil, 2020-11-10:
Thanks a lot for this bug report @PaulOKCrevinn. I've merged the code change from your colleague in #1180, so this should be resolved for now.

We unfortunately do not have access to Cadence tools (even less the ability to run them in CI), so I don't really know how we can ensure that Xcelium continues to work. I don't think any of our linters are able to detect this problem currently, but I've filed https://github.com/google/verible/issues/585 to get support to Verible lint added. Please add any additional information you might have to that issue.

closed 2020-11-10 by imphil

sriyerg, 2020-11-10:
It is strange to attach a label to `fork`. 

> Thanks a lot for this bug report @PaulOKCrevinn. I've merged the code change from your colleague in #1180, so this should be resolved for now.
> 
> We unfortunately do not have access to Cadence tools (even less the ability to run them in CI), so I don't really know how we can ensure that Xcelium continues to work. I don't think any of our linters are able to detect this problem currently, but I've filed [google/verible#585](https://github.com/google/verible/issues/585) to get support to Verible lint added. Please add any additional information you might have to that issue.

+@udinator - we should really get Ibex working with DVSim so that it can be run with any simulator.

imphil, 2020-11-10:
@sriyerg Ibex DV can already be executed with simulators other than VCS, dvsim support isn't a blocker there. It's just that we don't regularly use anything other than VCS.

sriyerg, 2020-11-10:
To clarify: 

```systemverilog 
fork: label
  begin: thread1
  end
  ...
join
disable label; // This is incorrect. 
disable thread1; // This is very much valid.
```

CathalMCrevinn, 2020-11-10:
Using `disable thread1;` as you're describing here will get you some messy differences between VCS and Xcelium. 
For example:
````
module test;
  initial begin
    fork
      begin: thread1
        task1();
      end
    join
    #10 disable thread1;
  end
  
  task task1();
    fork
      #20 $display("task1 print"); // Printed by Xcelium, not printed by VCS
    join_none
  endtask
endmodule
````

sriyerg, 2020-11-11:
> Using `disable thread1;` as you're describing here will get you some messy differences between VCS and Xcelium.
> For example:
> 
I would expect VCS and Xcelium (and all other simulators for that matter) to behave identically in this example, otherwise VCS would be violating the SV 1800-2017 spec! Section 9.6.2 clearly explains `disable thread1` should kill the execution of the code block, but not the children spawned by the thread (at any time, all the way from its epoch). This is useful if you don't want to be killing off threads spawned before the fork. Modifying your example a bit:

```systemverilog
module test;
  initial begin
    task1();

    fork
      begin: thread1
        // do some work
      end
    join_none
    #10 disable fork;   
  end
  
  task task1();
    fork
      #20 $display("task1 print"); // Gets printed with "disable thread1", but not with "disable fork"
    join_none
  endtask
endmodule
```

It is a bit counter-intuitive that `disable fork` kills off `task1()` as well, often resulting in some debug overhead. This is typically solved by wrapping it in an isolation fork (which I agree that is the most robust solution for this). However, `disable thread1` also works, if it is known that `thread1` does not spawn additional children. IMO rather than restricting our style guide by banning `disable thread1` type constructs altogether, it would be better to educate the user on their pros and cons.

PaulOKCrevinn, 2020-11-11:
Section 9.6.2 of the LRM deals with disable of tasks via labels. 
Section 9.6.3 of the LRM deals with disable of forks. 
I think that the compatibility issues arise between the simulators when code is written that mix these two disable types. 
We've seen this happen previously with other testbenches being ported from VCS to Xcelium, so we knew what to look for. 

It's safest in my opinion to avoid disabling of fork threads and only use "disable fork",