lowRISC/ibex #1175: assert code style to support formality tool

zjuwlh, 2020-11-05:
Hi,
when we run formality , the tool will report errors about the assertion related logic in RTL design, how to fix this issue without commenting these assertions in design?

Thanks,
Lihua

GregAC, 2020-11-05:
We'd expect no errors with the assertions. I'd need to see the errors in order to propose a fix. I don't think we're doing anything particular odd with the syntax so I'm a little surprised formality doesn't like them. It may be our assertion macros are doing things in a certain way it doesn't like and just need a bit of tweaking.

zjuwlh, 2020-11-06:
the formality tool will report errors such as the following:

**Syntax error found in the definition of argument macro `ASSERT_PULSE` in prim_assert.sv**

GregAC, 2020-11-06:
I'll need the full set of errors to fix anything, the `ASSERT_PULSE` macro has no issue with other tools so I cannot tell without more information on the errors what Formality's problem with that macro is.

zjuwlh, 2020-11-06:
it seem that formality can't support assign value to signal , such as high light in the following line.

define ASSERT(__name, __prop, **__clk = `ASSERT_DEFAULT_CLK, __rst = `ASSERT_DEFAULT_RST**)

and can pass if defining as the following in dummy macro.

define ASSERT(__name, __prop, __clk = , __rst =)

GregAC, 2020-11-11:
In that case you'll need to modify the assert macros to drop the use of the `__clk` and `__rst` arguments. I don't think we use them at all in Ibex so this shouldn't be hard to do, just create replacement files that provide the same assertion macros without `__clk` and `__rst` arguments replacing them with `ASSERT_DEFAULT_CLK` and `ASSERT_DEFAULT_RST` where they are used in the macros.

rswarbrick, 2021-01-25:
I'm closing this issue because we don't explicitly support the tool, and the problem comes from the fact that it seems formality doesn't correctly implement a SystemVerilog pre-processor.

closed 2021-01-25 by rswarbrick