lowRISC/ibex #1185: RV32EMC variant

subhajit26, 2020-11-10:
We are doing thesis on customization of Ibex core. We have implemented RV32IMC instructions successfully but going ahead we need to modify instruction set to RV32 EMC (16 registers), so can you please tell whether Ibex will support RV32EMC instructions.

tomroberts-lowrisc, 2020-11-10:
Hi @subhajit26 
Yes Ibex should support configuration, but it is not currently tested by any of our DV setups.

subhajit26, 2020-11-10:
1. Can you confirm whether this DV setup corresponds to the DV folder inside ibex repo or DV instruction generator inside vendor directory.
2. As I have already said that going ahead we will be using RV32EMC so can you please give me some advice on how to setup the Ibex configuration for RV32EMC . Basically I want to know where do we need to make modifications regarding Ibex core.

tomroberts-lowrisc, 2020-11-10:
1. Both I think - we never enable that configuration, and I don't believe the instruction generator supports it
2. I'm not sure if this is what you are asking, but the core is configured by setting parameters. So wherever you instantiate the core: set RV32E to 1 and RV32M to `ibex_pkg::RV32MFast`, `ibex_pkg::RV32MSingleCycle` or `ibex_pkg::RV32MSlow` depending on which multiplier you want. The C extension is always enabled in Ibex.

subhajit26, 2020-11-10:
As you have said that by instantiating the core we can change the variant may be RV32E or RV32I , so going ahead when I perform synthesis will the synthesis tool be able to recognize the difference in variants because both the variants have different register count. or will it give same result since I am sure how the configuration is done.

tomroberts-lowrisc, 2020-11-11:
I'm not sure I understand the question, but if you synthesize both configurations you should see a fairly big area difference (the E extension removes half of the register file).

rswarbrick, 2021-01-25:
I'm closing this issue because I believe the question has been answered. Please re-open or file a new issue if we've missed something.

closed 2021-01-25 by rswarbrick