lowRISC/ibex #1186: C directed test are not running

rahulraveendran15-coder, 2020-11-11:
Hi,
I just want to run a sample hello world program with ibex core. But the DV instruction generator is throwing error during compilation.
The run command I tried is as below from DV  directory:
`run --c_tests ../riscv-ovpsim/examples/hello/hello_world.c --simulator xlm`

My hello world program is as of below:
```
#include <stdio.h>
int main() {
   // printf() displays the string inside quotation
   printf("Hello, World! \n");
   return 0;
}

```

Error I got is something like undefined reference to puts, it is shown below:

```
Wed, 11 Nov 2020 08:49:41 INFO     Creating output directory: out_2020-11-11
Wed, 11 Nov 2020 08:49:41 INFO     Compiling c test : ../riscv-ovpsim/examples/hello/hello_world.c
/home/rraveendran/.local/opt/riscv32/riscv-gnu-toolchain/lib/gcc/riscv32-unknown-elf/10.2.0/../../../../riscv32-unknown-elf/bin/ld: warning: cannot find entry symbol _start; defaulting to 0000000080000000
/home/rraveendran/.local/opt/riscv32/riscv-gnu-toolchain/lib/gcc/riscv32-unknown-elf/10.2.0/../../../../riscv32-unknown-elf/bin/ld: /tmp/ccFjtyi3.o: in function `.L0 ':
hello_world.c:(.text+0x10): undefined reference to `puts'
collect2: error: ld returned 1 exit status
Traceback (most recent call last):
  File "/home/rraveendran/.local/bin/run", line 11, in <module>
    load_entry_point('riscv-dv', 'console_scripts', 'run')()
  File "/home/rraveendran/thesis_ibex/riscv-dv/run.py", line 1020, in main
    args.debug)
  File "/home/rraveendran/thesis_ibex/riscv-dv/run.py", line 575, in run_c
    run_cmd_output(cmd.split(), debug_cmd=debug_cmd)
  File "/home/rraveendran/thesis_ibex/riscv-dv/scripts/lib.py", line 195, in run_cmd_output
    raise exc
  File "/home/rraveendran/thesis_ibex/riscv-dv/scripts/lib.py", line 192, in run_cmd_output
    output = subprocess.check_output(cmd)
  File "/usr/lib64/python3.6/subprocess.py", line 356, in check_output
    **kwargs).stdout
  File "/usr/lib64/python3.6/subprocess.py", line 438, in run
    output=stdout, stderr=stderr)
subprocess.CalledProcessError: Command '['/home/rraveendran/.local/opt/riscv32/riscv-gnu-toolchain//bin/riscv32-unknown-elf-gcc', '-mcmodel=medany', '-nostdlib', '-nostartfiles', '../riscv-ovpsim/examples/hello/hello_world.c', '-I/home/rraveendran/thesis_ibex/riscv-dv/user_extension', '-T/home/rraveendran/thesis_ibex/riscv-dv/scripts/link.ld', '-o', 'out_2020-11-11/directed_c_tests/hello_world.o', '-march=rv32imc', '-mabi=ilp32']' returned non-zero exit status 1.
```

Can any one please help me on this, I tried adding the same c test in test list yaml file and the same error is getting.

GregAC, 2020-11-11:
As far as I know we don't use C directed tests in the Ibex DV environment. You may be better off opening an issue in the RISC-V DV repository for support on this feature.

However I would note your use of `printf`. We do not provide a C standard library for any Ibex environment in the repository so if you want to use `printf` and other features of the C standard library you will need to provide your own implementations.

rahulraveendran15-coder, 2020-11-11:
I have some reference algorithms in c to run on ibex core and measure its performance.
So before that I was planning to run a sample hello wolrd on ibex core and check it with whole DV flow.
If I am wrong could you please suggest which approach should I make for my work?

(1) As you said  >>  We do not provide a C standard library for any Ibex environment in the repository so if you want to use printf and other features of the C standard library you will need to provide your own implementations >> what should I need to use instead of printf to get rid of this error?
Also what you meant by providing your own implemenations>> How can we add this ?

GregAC, 2020-11-11:
I'd recommend using simple system to trial C code on the Ibex core: https://github.com/lowRISC/ibex/tree/master/examples/simple_system

The DV environment is for verification, not performance analysis so isn't suitable for that task (e.g. it will have randomised delays on memory accesses).

> what should I need to use instead of printf to get rid of this error?

You need to avoid using anything from the C standard library, simple system provides a very minimal set of functions to output things, see: https://github.com/lowRISC/ibex/blob/master/examples/sw/simple_system/common/simple_system_common.h

I'd note there is a printf implementation provided as part of Coremark here: https://github.com/lowRISC/ibex/blob/master/examples/sw/benchmarks/coremark/ibex/ee_printf.c you could take that file and build it in with your software and it will work on simple system (it will need some minor modifications to remove the coremark header include though).

>  How can we add this ?

A full C standard library implementation isn't trivial, you could try porting something like https://uclibc-ng.org/ to Ibex simple system if you need one but this isn't something I can explain how to do in a github issue.

rahulraveendran15-coder, 2020-11-13:
> I'd note there is a printf implementation provided as part of Coremark here: https://github.com/lowRISC/ibex/blob/master/examples/sw/benchmarks/coremark/ibex/ee_printf.c you could take that file and [...]

I moved the file into /home/rraveendran/thesis_ibex/ibex/examples/sw/simple_system/common and changed the format into ".h" also moved the necessary header files to the same path and added the ee_printf.h inside hello_test.c, now it is showing some undefined references error as below: But I can see that these are defined in the same file.
```
/home/rraveendran/.local/opt/riscv32/riscv-gnu-toolchain/lib/gcc/riscv32-unknown-elf/10.2.0/../../../../riscv32-unknown-elf/bin/ld: hello_test.o: in function `flt':
/home/rraveendran/thesis_ibex/ibex/examples/sw/simple_system/common/ee_printf.h:340: undefined reference to `__ltdf2'
/home/rraveendran/.local/opt/riscv32/riscv-gnu-toolchain/lib/gcc/riscv32-unknown-elf/10.2.0/../../../../riscv32-unknown-elf/bin/ld: hello_test.o: in function `parse_float':
/home/rraveendran/thesis_ibex/ibex/examples/sw/simple_system/common/ee_printf.h:253: undefined reference to `fcvtbuf'

```

GregAC, 2020-11-13:
We're moving out of the scope of support we can offer via this channel (it's for issues directly related to Ibex and the associated things provided in the repository, this is more of a general C programming question at this point) but I can give you some hints.

If you look at the error the reference to `parse_float` should give you pause for thought. Ibex doesn't have float support. Soft-float support (where the compiler emits instructions to emulate hardware floating point in software) can be utilised but in this case doesn't look to be working properly (it's trying to call float related functions that don't exist). 

If you take a look at the ee_printf.c file you'll see the use of a `HAS_FLOAT` define around the floating point stuff. Try running a build without that being defined and see if that fixes things.

rahulraveendran15-coder, 2020-11-22:
@GregAC 
I am able to see ```
CFLAGS ?= -march=$(ARCH) -mabi=ilp32 -static -mcmodel=medany -Wall -g -Os\
	-fvisibility=hidden -nostdlib -nostartfiles -ffreestanding $(PROGRAM_CFLAGS)
```
in common.mk file where the -nostdlib option is passed,  when I removed that option and ran a test with printf in c code the ibex system is going to simple_exc_handler. Why is it  behaving like this?

rswarbrick, 2020-11-23:
Hi Rahul,

As Greg said, this is getting a bit beyond what makes sense to discuss here. I think that maybe you need to do some reading about the difference between a [hosted and a freestanding C implementation](https://en.cppreference.com/w/cpp/freestanding). As a low-level embedded target, Ibex is freestanding.

closed 2021-01-07 by imphil