lowRISC/ibex #1187: Can we run the simple system of hello using xcelium simulator

rahulraveendran15-coder, 2020-11-11:
Hi, 
Is it possible to run the simple system using xcelium simulator?  I can see the support for vcs, verialtor and rivera. But nothing related to xcelium. Is it possble if not possible which all files do I need to update for adding support to xcelium?

`https://github.com/lowRISC/ibex/tree/master/examples/simple_system`

GregAC, 2020-11-11:
There's not any xcelium support currently but it shouldn't be too hard to get it working. First step would be to add xcelium to the tools in simple system fusesoc core file: https://github.com/lowRISC/ibex/blob/master/examples/simple_system/ibex_simple_system.core

I'd suggest taking a look at the fusesoc documentation to see how it all works: https://fusesoc.readthedocs.io/en/rtd/fusesoc.html

imphil, 2020-11-11:
You can try the default config with 

```
fusesoc --cores-root=. run --target=sim --tool=xcelium --setup --build lowrisc:ibex:ibex_simple_system --RV32E=0 --RV32M=ibex_pkg::RV32MFast
```

That should work if Xcelium is fine with the default configuration. If not, look at the error message and go from there.

rahulraveendran15-coder, 2020-11-18:
@imphil  when I tried the command I got the below warnings
WARNING: util/check_tool_requirements.py has unknown file type ''
WARNING: tool_requirements.py has unknown file type ''
Do I need to add xcelium in this file?
 Also  I am not seeing any build is happening when trying with xcelium. Do I need to add anything in the ibex_simple_system.core file inside examples directory also anything like` --SRAMInitFile=` in vcs command?
Under build directory I am getting these files only inside sim-xcelium.
[image]

imphil, 2020-11-18:
The warnings are fine, you can ignore them.

If you don't get any other error, the simulation was most likely built properly and you can now run it. Have a look at the Makefile in the sim-xcelium directory (or pass `--run` after `--build` to fusesoc to run XCelium directly for you).

But there's no way to tell for me for sure: please *always* include full logs of your run (you can attach files to GitHub issues!).

rahulraveendran15-coder, 2020-11-18:
Thank you very much I am seeing no VPI_MODULES inside the make file.
Pleas
[generated_makefile.log](https://github.com/lowRISC/ibex/files/5559058/generated_makefile.log)

e find the attachemnt log. 
Also do I need to add --SRAMInitFile=<path to .vmem file> when using xcelium(like in vcs)?

imphil, 2020-11-18:
so what happens if you run "make run-gui"?

rahulraveendran15-coder, 2020-11-18:
It is throwing some elabopration errors i nthe rtl/simple_system and also showing the warning Parameter cannot be overwritten through command line 
[xrun.log](https://github.com/lowRISC/ibex/files/5559108/xrun.log)
Log is attached.
[run-gui.log](https://github.com/lowRISC/ibex/files/5559104/run-gui.log)

rahulraveendran15-coder, 2020-11-18:
> The warnings are fine, you can ignore them.
>
> If you don't get any other error, the simulation was most likely built properly and you can now run it. Have a look at the Makefile in the sim-xcelium [...]

closed 2020-11-18 by rahulraveendran15-coder

imphil, 2020-11-18:
The error you got is
```
xmelab: *E,DLYSYN (../src/lowrisc_ibex_ibex_simple_system_0/rtl/ibex_simple_system.sv,121|6): Delay illegal within an always_comb construct.
      #1 clk_sys = 1'b1;
      |
xmelab: *E,DLYSYN (../src/lowrisc_ibex_ibex_simple_system_0/rtl/ibex_simple_system.sv,122|6): Delay illegal within an always_comb construct.
xrun: *E,ELBERR: Error during elaboration (status 1), exiting.
```

The code in question isn't there any more. Please update to the latest master version of Ibex.

rahulraveendran15-coder, 2020-11-18:
Even updation also some elaboration error persists from src/simulator_ctrl.sv.
Also the defparams are not overriding from command line as I can see as warning in log.
[xrun.log](https://github.com/lowRISC/ibex/files/5559173/xrun.log)

imphil, 2020-11-18:
Looks like Xcelium is (rightfully so) more strict about some constructs than other simulators. Can you give it a try and fix the errors you see in the Xcelium log do a pull request? Compared to the work you're probably planning with Ibex, these should be easy fixes.

rahulraveendran15-coder, 2020-11-18:
Okay will check it out, but the problem is that the src folder and related files are created automatically when using fusesoc command so where should I need to fix it directly in the generated file or else I don't know from where and how it is generated.
If I am hardcoding the fix inside src directly then I need to run make run from inside sim-xcelium otherwise the fusesoc command will not work I guess or else the src will be built again with new set of files.

Also some more concerns I have :
(1) Why do I need give --run along with fusesoc command to run with xcelium? why only $(VPI_MODULES) is there in "all" of Makefile
 all: $(VPI_MODULES)

(2) How to simulate it after building? Or is it like xrun command all compilation elaboration and simulation will take place along with run in make file(not in gui mode)
(3)Also do I need to add --SRAMInitFile=<path to .vmem file> when using xcelium(like in vcs)?

imphil, 2020-11-18:
You always need to make the fixes in the source tree that you checked out, not in the `src` directory that FuseSoC created. After you made your changes, you can either re-run fusesoc (recommended), or pass `--no-export` to FuseSoC once to prevent FuseSoC from copying files.

1. There are three stages in building, setup, build, and run. The simulation is only run if you actually pass `--run`, or manually run Xcelium (e.g. through the Makefile).
2. VPI_MODULES can be empty if there are no VPI modules (as it is the case here).
3. See 1
4. Yes, if you want to run software.

imphil, 2020-11-18:
I also filed https://github.com/lowRISC/ibex/issues/1196 to keep track of the warnings we see in Xcelium. Please close this issue once you're done with your pull request.

rahulraveendran15-coder, 2020-11-18:
Okay I am checking on the issue and will update.
Regarding (1) >> is the --run only for xcelium since I am not seeing for VERILATOR and VCS(there for building and setup its one command and for running simulator its different)
(3) in Verilator is it possible to run all these building, setup and simulation in single step like in xcelium(Right now it is using separate commands for building and simulation)
(4) I am still not clear on this what is it, 

(a) Because when using verilator I am not seeing any `--SRAMInitFile` along with fusesoc command but I am seeing we are giving "-`-meminit=ram,<sw_elf_file>"` the elf file generated along while running the simulator

(b) In VCS the `.vmem` is given along with fusesoc while building and no elf  given when running the simulator >> why is it so?

(c) In xcelium so which one should I mention` .vmem `file or `.elf` file?

(d) Why for verilator .elf is required and why .vem is required for vcs even both are added in different stages(for verialtor its ins simulation step for vcs its in simulation step)?

Fresher14, 2020-11-23:
[image]

It took some time but at last it works with Xcelium (hello_test).

rahulraveendran15-coder, 2020-11-23:
@Fresher14 
If possible could you please  share the command you used with fusesoc to build and run the gui?
I just want to compare with my approach
I used the below command with Xcelium to run :+1: 
fusesoc --cores-root=. run --target=sim --tool=xcelium --setup --build --run lowrisc:ibex:ibex_simple_system --RV32E=0 --RV32M=ibex_pkg::RV32MFast --ibex_simple_system.SRAMInitFile=\\\"/home/rraveendran/thesis_ibex/ibex/examples/sw/simple_system/hello_test/hello_test.vmem\\\"

For Gui mode I need to run the above command without --run and then run the make run-gui from the sim-xcelium directory. Do you also follow the same approach to get the above attached waveform.

Fresher14, 2020-11-23:
Then environment I have access to is highly controlled, with very low flexibility. Therefore I cannot use fusesoc/verilator. Here is my script, that calls xrun directly (sw is pre-compiled in ../examples/sw/simple_system/hello_test).

[readme_run_simple_system.txt](https://github.com/lowRISC/ibex/files/5584291/readme_run_simple_system.txt)