lowRISC/ibex #1188: How to understand the command and usage of fusesoc in simple system

rahulraveendran15-coder, 2020-11-11:
Hi,
I just want to know how the usage of the fusesoc command for building the simulation.

`fusesoc --cores-root=. run --target=sim --setup --build lowrisc:ibex:ibex_simple_system --RV32E=0 --RV32M=ibex_pkg::RV32MFast
`
(1) Is there any documentation on how to use this and when to use even I am not seeing any script to refer
(2) Also in this command its given as RV32E=0, but as I can see in the RTL  the RV32E is defined as a parameter which cannot be overridden by command line as per systemverilog construct. So how this --RV32E=1 or --RV32E=0 changing the parameter value inside core_tracing or simple_system later on bypassed to inner modules?

tomroberts-lowrisc, 2020-11-11:
1) The documentation for fusesoc is here: https://fusesoc.readthedocs.io/en/rtd/index.html
2) When you pass options like `--RV32E=1` to fusesoc, it will pass those parameter overrides on to the tool in whatever format that tool requires (for example this results in `-pvalue+ibex_simple_system.RV32E=1` being added to the run script when vcs is invoked)

rswarbrick, 2021-01-25:
I'm closing this issue because I believe Tom answered the question above. Please re-open or open a new issue if we've missed something.

closed 2021-01-25 by rswarbrick