lowRISC/ibex #1189: cpu can't wake up after executing wfi to go into sleep mode

zjuwlh, 2020-11-12:
Hi ,

When running riscv_debug_instr_test, fo r some seed(SEED=16926 ITERATIONS=25), cpu will execute wfi and go into sleep during non-debug mode, in this case, if the return value of decode_instr is always 0, debug_req isn't asserted in DV test bench, the cpu will always is in sleep mode and can't wake up. does anyone also find this issue? and how to fix in test bench

tomroberts-lowrisc, 2020-11-12:
Probably one for @udinator

rswarbrick, 2021-04-05:
@udinator: Ping!

udinator, 2021-04-09:
I think @GregAC has fixed this in one of his previous bug-fix PRs (#1266 I think?) - could you confirm?

GregAC, 2021-04-12:
Yes that should have done it, so I'll close this @zjuwlh do let us know if you see a similar failure again with the fix. I haven't been seeing those failures myself in regression runs I've been doing.

closed 2021-04-12 by GregAC