lowRISC/ibex #1193: syn_yosys.sh failure at 'translate_timing_csv.py' phase

YanekJ, 2020-11-16:
Hi,
Not sure if this is related to an environment, python setup or anything else but I'm getting below error towards the end of syn_yosys.sh script execution (yosys completed, sta completed - conversion of timing results failed):

Replacing existing blackbox module XOR2_X1.
Replacing existing blackbox module XOR2_X2.
Imported 134 cell types from liberty file.

3. Executing Verilog-2005 frontend: ./syn_out/ibex_16_11_2020_16_16_55/generated/ibex_core_netlist.v
Parsing Verilog input from `./syn_out/ibex_16_11_2020_16_16_55/generated/ibex_core_netlist.v' to AST representation.
Generating RTLIL representation for module `\ibex_core'.
Successfully finished Verilog frontend.

End of script. Logfile hash: 74b8848413, CPU: user 2.85s system 0.10s, MEM: 201.43 MB peak
Yosys 0.9+3675 (git sha1 71ca9a82, gcc 7.5.0-3ubuntu1~18.04 -fPIC -Os)
Time spent: 88% 2x read_verilog (2 sec), 11% 4x read_liberty (0 sec), ...
**Traceback (most recent call last):
  File "./python/translate_timing_csv.py", line 41, in <module>
    main()
  File "./python/translate_timing_csv.py", line 28, in main
    path_info, '{}/{}'.format(args.gen_out, ys_translated_names))
  File "/home/yanek/Sources/riscv/ibex/syn/python/flow_utils.py", line 69, in translate_path_info
    translated_names_dict = build_translated_names_dict(translated_names_file)
  File "/home/yanek/Sources/riscv/ibex/syn/python/flow_utils.py", line 54, in build_translated_names_dict
    translated_names = open(translated_names_file)
FileNotFoundError: [Errno 2] No such file or directory: './syn_out/ibex_16_11_2020_16_16_55/generated/ys_translated_names'
Traceback (most recent call last):
  File "./python/translate_timing_csv.py", line 41, in <module>
    main()
  File "./python/translate_timing_csv.py", line 28, in main
    path_info, '{}/{}'.format(args.gen_out, ys_translated_names))
  File "/home/yanek/Sources/riscv/ibex/syn/python/flow_utils.py", line 69, in translate_path_info
    translated_names_dict = build_translated_names_dict(translated_names_file)
  File "/home/yanek/Sources/riscv/ibex/syn/python/flow_utils.py", line 54, in build_translated_names_dict
    translated_names = open(translated_names_file)
FileNotFoundError: [Errno 2] No such file or directory: './syn_out/ibex_16_11_2020_16_16_55/generated/ys_translated_names'
Traceback (most recent call last):
  File "./python/translate_timing_csv.py", line 41, in <module>
    main()
  File "./python/translate_timing_csv.py", line 28, in main
    path_info, '{}/{}'.format(args.gen_out, ys_translated_names))
  File "/home/yanek/Sources/riscv/ibex/syn/python/flow_utils.py", line 69, in translate_path_info
    translated_names_dict = build_translated_names_dict(translated_names_file)
  File "/home/yanek/Sources/riscv/ibex/syn/python/flow_utils.py", line 54, in build_translated_names_dict
    translated_names = open(translated_names_file)
FileNotFoundError: [Errno 2] No such file or directory: './syn_out/ibex_16_11_2020_16_16_55/generated/ys_translated_names'
Traceback (most recent call last):
  File "./python/translate_timing_csv.py", line 41, in <module>
    main()
  File "./python/translate_timing_csv.py", line 28, in main
    path_info, '{}/{}'.format(args.gen_out, ys_translated_names))
  File "/home/yanek/Sources/riscv/ibex/syn/python/flow_utils.py", line 69, in translate_path_info
    translated_names_dict = build_translated_names_dict(translated_names_file)
  File "/home/yanek/Sources/riscv/ibex/syn/python/flow_utils.py", line 54, in build_translated_names_dict
    translated_names = open(translated_names_file)
FileNotFoundError: [Errno 2] No such file or directory: './syn_out/ibex_16_11_2020_16_16_55/generated/ys_translated_names'
Traceback (most recent call last):
  File "./python/translate_timing_csv.py", line 41, in <module>
    main()
  File "./python/translate_timing_csv.py", line 28, in main
    path_info, '{}/{}'.format(args.gen_out, ys_translated_names))
  File "/home/yanek/Sources/riscv/ibex/syn/python/flow_utils.py", line 69, in translate_path_info
    translated_names_dict = build_translated_names_dict(translated_names_file)
  File "/home/yanek/Sources/riscv/ibex/syn/python/flow_utils.py", line 54, in build_translated_names_dict
    translated_names = open(translated_names_file)
FileNotFoundError: [Errno 2] No such file or directory: './syn_out/ibex_16_11_2020_16_16_55/generated/ys_translated_names'**
Area in kGE =  26.37

rswarbrick, 2020-11-16:
@GregAC: I think this script was yours? Can you help Yanek work out what went wrong?

GregAC, 2020-11-16:
The synthesis flow is unfortunately a bit fragile. We've not yet got it running in CI and it's not always used so can break without us noticing sometimes.

I can't directly tell what is wrong here. The thing to check is what `translate_timing_rpts.sh` is up to (https://github.com/lowRISC/ibex/blob/master/syn/translate_timing_rpts.sh).

It first runs `build_translated_names.py` this will load the timing reports to get some signal names, build a yosys script `translate_names.ys` and run it in yosys. Then `translate_timing_csv.py` takes the yosys output which gives you the human readable signal names and transforms the CSV.

I suspect it's the first step that's going wrong somewhere. Check the `generated` directory in the synthesis output directory. It you can't see both `translate_names.ys` and `ys_translated_names` in there (checking they have contents, expect around 2000 lines) then `build_translated_names.py` has failed.

One reason for `build_translated_names.py` to fail would be the STA run hasn't actually produced any timing reports for it to find signal names in, check the `reports/timing` directory and see if it has. If it hasn't take a look at `log/sta.log` and see if there's any obvious issues in the STA run.

YanekJ, 2020-11-16:
I checked and 'generated' directory has translate_names.ys file but with below contents (I think 'select' commands are missing):

tcl ./tcl/yosys_post_synth.tcl
cd ibex_core

reports/timings directory has:

in2out.csv.rpt
in2out.rpt
in2reg.csv.rpt
in2reg.rpt
overall.csv.rpt
overall.rpt
reg2out.csv.rpt
reg2out.rpt
reg2reg.csv.rpt
reg2reg.rpt

GregAC, 2020-11-16:
Is there anything in the report files? The empty `translate_names.ys` would indicate `build_translated_names.py` has failed to pick up any signal names from the timing reports.

I've just the synthesis locally on my machine and it's working ok there.

What library are you using? One possibility is OpenSTA failed to load it or couldn't pull any timing information from it so failed to produce any reports.

YanekJ, 2020-11-16:
Report files are populated alright, few top lines from example

$ cat ibex_16_11_2020_16_16_55/reports/timing/in2out.csv.rpt 
irq_fast_i[7],instr_addr_o[27],1.2074
irq_fast_i[7],instr_addr_o[30],1.2155
irq_fast_i[7],instr_addr_o[29],1.2159
irq_fast_i[7],instr_addr_o[25],1.2174
irq_fast_i[7],instr_addr_o[28],1.2198
irq_fast_i[7],instr_addr_o[26],1.2236

YanekJ, 2020-11-16:
I think it has to be some shell/python interfacing issue. Even if I execute below (after sourcing env script, and commenting-out other lines from translate_timing_rpts.sh script):

./python/build_translated_names.py ibex_core ./$LR_SYNTH_OUT_DIR/generated ./$LR_SYNTH_OUT_DIR/reports/timing/in2reg.csv.rpt

I'm getting this truncated translate_names.ys and no ys_translated_names file

$ cat syn_out/ibex_16_11_2020_16_16_55/generated/translate_names.ys 
tcl ./tcl/yosys_post_synth.tcl
cd ibex_core

YanekJ, 2020-11-16:
I suspect the regex in flow_utils.py can't match any cell names generated by sta tool. My sta version is 2.2.0.

GregAC, 2020-11-16:
I think you'll need to debug the `build_translated_names.py` file. It's just going through the CSVs and pulling out signal names some aspect of that is failing. I'd start by adding some print statements to `extract_path_info` in `syn/python/flow_utils.py` which opens the CSV and pulls out the path info.

GregAC, 2020-11-16:
> I suspect the regex in flow_utils.py can't match any cell names generated by sta tool. My sta version is 2.2.0.

Ah that may be it, I'm using 2.0.18 guess this has changed recently.

YanekJ, 2020-11-16:
Maybe the regex "generated_cell_re" is too strict. Shouldn't we get the full word from the first csv column and then remove square brackets from signal arrays? Maybe you have better idea,

GregAC, 2020-11-16:
There's an issue with the reports themselves with OpenSTA. I just tried a run with 2.2.0 and got this in reg2reg.csv.rpt:

```
Q,D,1.3626
Q,D,1.3626
Q,D,1.3626
Q,D,1.3626
Q,D,1.3626
Q,D,1.3626
Q,D,1.3626
Q,D,1.3626
Q,D,1.3626
Q,D,1.3626
Q,D,1.3626
Q,D,1.3626
Q,D,1.3626
Q,D,1.3626
Q,D,1.3626
Q,D,1.3626
Q,D,1.3626
Q,D,1.3626
Q,D,1.3626
Q,D,1.3626
Q,D,1.3626
Q,D,1.3626
Q,D,1.3626
Q,D,1.3626
Q,D,1.3626
Q,D,1.3626
Q,D,1.3626
Q,D,1.3626
Q,D,1.3626
Q,D,1.3626
```

Previously this contained the cell names this script was attempting to translate.

So a full fix needs some adjustment to the STA flow also.

YanekJ, 2020-11-16:
Agreed, the contents of overall.csv.rpt are suspicious too:

Q,D,0.0000
Q,D,0.0000
Q,D,0.0000
Q,D,0.0000
Q,D,0.0000
Q,D,0.0000

PR #1241 (2021-01-19): [syn] Fix timing reports in synthesis flow
This fixes a bug where .csv.rpt files weren't being generated properly.
OpenSTA was outputting a simple 'Q' or 'D' for start and end path points
where the full instance name was required (which is then fed to yosys to
translate to a human readable name). The issue was in the `timing_report`
proc in syn/tcl/sta_utils.tcl which needed to request the full name of
the start and end points.

The syn README is updated to note which tool versions have been used to
test the flow.

Fixes #1193
commit 75e1c281 (2021-01-19): [syn] Fix timing reports in synthesis flow

This fixes a bug where .csv.rpt files weren't being generated properly.
OpenSTA was outputting a simple 'Q' or 'D' for start and end path points
where the full instance name was required (which is then fed to yosys to
translate to a human readable name). The issue was in the `timing_report`
proc in syn/tcl/sta_utils.tcl which needed to request the full name of
the start and end points.

The syn README is updated to note which tool versions have been used to
test the flow.

Fixes #1193
closed 2021-01-19 by GregAC