lowRISC/ibex #1194: [syn] Add more robust error detection/reporting to synthesis flow

GregAC, 2020-11-16:
The synthesis flow uses a variety of python, shell and TCL scripts that expect certain things to be correct. When it breaks the errors it produces don't obviously put to the problem (as illustrated by https://github.com/lowRISC/ibex/issues/1193)

More thought should be put into detecting issues at the point they occur in the flow so sensible error reporting can occur.

rswarbrick, 2021-01-27:
@GregAC: What concrete steps do you think need taking before we can close this issue? At the moment, it'll probably remain open forever, which is not massively helpful.

GregAC, 2021-01-28:
I am looking at it as part of the work I'm doing around timing/feedthroughs.

Two main things to look at:

1. Add better error handling to the bits of python (not sure there's any at the moment)
2. Look into what error handling can be added to the tcl bits. Yosys/OpenSTA may just exit with an error code rather than allowing the tcl to do anything (or just write to stderr but command completes with no indications of problems on the TCL side) but I need to work that out.