lowRISC/ibex #1196: Fix Xcelium warnings and errors with simple system

imphil, 2020-11-18:
```
      .test_en_i             ('b0),
                                |
xmelab: *W,CUVMPW (../src/lowrisc_ibex_ibex_simple_system_0/rtl/ibex_simple_system.sv,185|32): port sizes differ in port connection (32/1).
      .debug_req_i           ('b0),
                                |
xmelab: *W,CUVMPW (../src/lowrisc_ibex_ibex_simple_system_0/rtl/ibex_simple_system.sv,214|32): port sizes differ in port connection (32/1).
      .fetch_enable_i        ('b1),
                                |
xmelab: *W,CUVMPW (../src/lowrisc_ibex_ibex_simple_system_0/rtl/ibex_simple_system.sv,216|32): port sizes differ in port connection (32/1).
  always_ff @(posedge clk_i or negedge rst_ni) begin
          |
xmelab: *E,MULAXX (../src/lowrisc_ibex_sim_shared_0/rtl/sim/simulator_ctrl.sv,59|10): Multiple drivers to always_ff output variable sim_finish detected.
  always_ff @(posedge clk_i or negedge rst_ni) begin
          |
xmelab: *E,MULAXX (../src/lowrisc_ibex_sim_shared_0/rtl/sim/simulator_ctrl.sv,59|10): Multiple drivers to always_ff output variable sim_finish detected.
  always_ff @(posedge clk_i or negedge rst_ni) begin
          |
xmelab: *E,MULAXX (../src/lowrisc_ibex_sim_shared_0/rtl/sim/simulator_ctrl.sv,59|10): Multiple drivers to always_ff output variable sim_finish detected.
xrun: *E,ELBERR: Error during elaboration (status 1), exiting.
```

rahulraveendran15-coder, 2020-11-18:
@imphil 
I have solved the multiple driver issue and also when using xcelium when overriding parameter from command line it has to use the full hierarchical path like for example "ibex_simple_system.SRAMInitFile" inside ibex_simple_system.core file(The other parameters I havent changed now because the default value is matching in sv file).
I have modified it for testing and tried running the below command:
`fusesoc --cores-root=. run --target=sim --tool=xcelium --setup --build --run lowrisc:ibex:ibex_simple_system --RV32E=0 --RV32M=ibex_pkg::RV32MFast --ibex_simple_system.SRAMInitFile=~/thesis_ibex/ibex/examples/sw/simple_system/hello_test/hello_test.vmem`

But this is showing some assertion failure related to memory/instruction from the file "/rtl/ibex_compressed_decoder.sv" line 287.

Could you please confirm my run command is correct or not or memory loading is correct or not.
Because when I try the above command with `--run-gui` instead of `--run` it throwed error as unknown command. Also as I mentioned earlier for verilator and VCS  I am seeing for building one command and for simulating another command. For xcelium does it need to be anything like this, right now I am using a single fusesoc command as you mentioned before.

Please see the error log for more details.
[xrun.log](https://github.com/lowRISC/ibex/files/5561655/xrun.log)

rahulraveendran15-coder, 2020-11-20:
@imphil 
I have solved the errors due to LRM violation that were causing with xcelium. Do I need to push the changes?

Also I have some doubts regarding the command usage:
Right now I am using the  below command to run simple_system and it is running fine:
`fusesoc --cores-root=. run --target=sim --tool=xcelium --setup --build --run lowrisc:ibex:ibex_simple_system --RV32E=0 --RV32M=ibex_pkg::RV32MFast --ibex_simple_system.SRAMInitFile=\\\"/home/rraveendran/thesis_ibex/ibex/examples/sw/simple_system/hello_test/hello_test.vmem\\\"`
(1)How can I run in gui mode along with fusesoc command (Not going to the sim-xcelium directory and giving `make run-gui`)
I tried the below command but it was showing error it cannot recognise --run-gui
`fusesoc --cores-root=. run --target=sim --tool=xcelium --setup --build --run-gui lowrisc:ibex:ibex_simple_system --RV32E=0 --RV32M=ibex_pkg::RV32MFast --ibex_simple_system.SRAMInitFile=\\\"/home/rraveendran/thesis_ibex/ibex/examples/sw/simple_system/hello_test/hello_test.vmem\\\"`
(2) One more doubt I have is that,
   why verilator and vcs have two separate commands for build and simulation? At the same time for Riviera-PRO it only single command for both build and simulation(I hope same case with Xcelium). Why is it so?
(3) Can I use the [-t] option along with xcelium to get the fst trace? Along with fusesoc command it is not able to recognise [-t]? From which file this [-t] switch is recognising
(4) what is the difference between [-t] and -t ? How can I use it with xcelium?
When I am giving [-t] alone no sim.fst is getting created
When I am giving -t alone (without [-t]) both sim.fst and the simulations statistics is displayed in terminal

imphil, 2020-11-20:
> I have solved the errors due to LRM violation that were causing with xcelium. Do I need to push the changes?

Yes, please do a pull request. You can find more information on how to do that in the GitHub documentation (e.g. at https://docs.github.com/en/free-pro-team@latest/github/collaborating-with-issues-and-pull-requests/creating-a-pull-request#changing-the-branch-range-and-destination-repository), and by Googling for "GitHub pull requests".

Fresher14, 2020-11-23:
@imphil  I get exactly same results.
Replacing 'always_ff' with with a simple 'always' was enough to pass hello_test in my environment.
On top of this, I had to retrieve 2 files generated by FuseSoc:
prim_clock_gating.sv
prim_ram_2p.sv
+ force a define RVFI
+ source dv/uvm/core_ibex/common/prim/prim_pkg.sv

rahulraveendran15-coder, 2020-11-23:
@Fresher14 
```
>> On top of this, I had to retrieve 2 files generated by FuseSoc:
prim_clock_gating.sv
prim_ram_2p.sv
Then force a define RVFI
Then source dv/uvm/core_ibex/common/prim/prim_pkg.sv
```

What you mean by this could you please explain? Because I didnt do any changes on these files.

Fresher14, 2020-11-23:
See topic #1187