lowRISC/ibex #1197: Command usage of simple system

rahulraveendran15-coder, 2020-11-21:
I have some doubts regarding the command usage of simple_system:

(1):
Right now I am using the below command to run simple_system and it is running fine:
`fusesoc --cores-root=. run --target=sim --tool=xcelium --setup --build --run lowrisc:ibex:ibex_simple_system --RV32E=0 --RV32M=ibex_pkg::RV32MFast --ibex_simple_system.SRAMInitFile=\\\"/home/rraveendran/thesis_ibex/ibex/examples/sw/simple_system/hello_test/hello_test.vmem\\\"`

(a) Does the `--run` command along with fusesoc is corresponding to the "run" in Makefile under sim-xcelium? if its correct then why `--run-gui` is not able to recognise?

(b) How can I run in gui mode along with fusesoc command (Not going to the sim-xcelium directory and giving make run-gui)
I tried the below command but it was showing error it cannot recognise `--run-gui`. 
`fusesoc --cores-root=. run --target=sim --tool=xcelium --setup --build --run-gui lowrisc:ibex:ibex_simple_system --RV32E=0 --RV32M=ibex_pkg::RV32MFast --ibex_simple_system.SRAMInitFile=\\\"/home/rraveendran/thesis_ibex/ibex/examples/sw/simple_system/hello_test/hello_test.vmem\\\"`

(2):
(a) why verilator and vcs have two separate commands for build and simulation? At the same time for Riviera-PRO it only single command for both build and simulation(I hope same case with Xcelium). Why is it so?

(3):
(a) Can I use the [-t] option along with xcelium to get the fst trace? Along with fusesoc command it is not able to recognise [-t]? From which file this [-t] switch is recognising

(b) what is the difference between [-t] and -t ? How can I use it with xcelium?
When I am giving [-t] alone no sim.fst is getting created
When I am giving -t alone (without [-t]) both sim.fst and the simulations statistics is displayed in terminal

rswarbrick, 2020-11-23:
Hi Rahul,

Yes, FuseSoc's `--run` flag tells it to run stuff (your simulation, in this case). This gets passed through to a tool called [Edalize](https://github.com/olofk/edalize) which actually knows how to run each EDA tool. The Edalize backend for xcelium defines a `run-gui` target, but I don't think that's plumbed into FuseSoc. Of course, once you've built your simulation, you can always invoke the Makefile directly (`make -C path/to/build/dir run-gui`).

About Verilator and VCS: FuseSoc splits everything into 3 phases: setup, build and run. For Verilator and VCS, this split is pretty obvious: "setup" grabs the source code; "build" compiles it to something executable (e.g. for Verilator, this means transforming verilog to C++ and then compiling the results); "run" runs the simulation. I'm afraid I don't know about Riviera, but a quick glance at the edalize backend makes me think it's probably similar. There's a FuseSoc [user guide](https://fusesoc.readthedocs.io/en/master/user/index.html) which explains this in more detail. If you need more help with FuseSoc, it's probably best to ask that project.

I don't believe that Xcelium supports the FST trace format. If you know the commands to pass to Xcelium to enable wave dumping (probably in SHM format), it looks like you can pass them to the generated Makefile with the `XSIM_OPTIONS` or `XRUN_OPTIONS` arguments. So something like `make -C path/to/build run XSIM_OPTIONS='my options here'`. As far as I know, FuseSoc doesn't have plumbing to do this directly (thank goodness it splits the `build` and `run` phases!). For the exact options, you'll need to look in the Xcelium manual.

Finally, about the square brackets. Have you got confused about the help text from simple system? The Unix convention is that an optional argument gets surrounded by square brackets to show that it's optional: you'll see this in every man page, for example. Passing `-t` to the Verilator `simple_system` binary enables wave dumping (I think it's 't' for 'trace'). The Xcelium version doesn't have the same C++ top-level, so you have to control it differently (as described in the previous paragraph).

rahulraveendran15-coder, 2020-11-23:
Hi rswarbrick,
Thank you for the explanation.
In the last point I understood it is linux convention, but both [-t] or -t along with our run command need to give same result right?

rswarbrick, 2020-11-23:
It's the same convention as every other command. If the help text says `my_prog [--some-arg]`, it means that you can either specify `--some-arg` or not. I've never seen literal square brackets expected as command line parameters, I don't think.

rswarbrick, 2021-01-27:
I'm closing this report, since I think the question has been answered.

closed 2021-01-27 by rswarbrick