lowRISC/ibex #1204: "no_wfi" is usually being set to 1, even for tests with "+no_wfi=0"

CathalMCrevinn, 2020-11-30:
In ``dv/uvm/core_ibex/riscv_dv_extension/testlist.yaml``, riscv_debug_wfi_test has ``+no_wfi=0`` as a gen_opt.
Example run:  ``"make TEST=riscv_debug_wfi_test SIMULATOR=xlm ITERATIONS=5 SEED=23293"``
For this example run, in ``instr_gen/sim_riscv_debug_wfi_test_0.log``, the configuration for the first iteration has no_wfi set to 0, but the configurations for the remaining four iterations have it set to 1. 
It's not just an issue of ``init_privileged_mode`` being USER_MODE in the configuration. Some iterations have it as MACHINE_MODE but with ``no_wfi`` still set to 1.
It's also not specific to riscv_debug_wfi_test, it applies to the other tests with ``+no_wfi=0`` too.

udinator, 2020-11-30:
Thanks for raising this issue!
This may be due to some unintended side effect of the latest commits in the vendored in generator, I will take a look.

rswarbrick, 2021-04-05:
@udinator: Any progress here?

udinator, 2021-04-09:
Hi @CathalMCrevinn I have still been unsuccessful in recreating this issue on both VCS and Xcelium simulators. I am seeing that if the core is booted into M-mode then `no_wfi` is constrained to 0, and is kept at 1 if U-mode is used (as intended).
Could you try syncing with the current upstream and trying again?

closed 2021-12-10 by GregAC